iberianpig/fusuma #225: Crash with Wacom CTH-480

cab404, 2021-03-30:
**Describe the bug**
Fusuma crashes when any gesture is used on Wacom CTH-480.

```
Traceback (most recent call last):
	26: from /nix/store/zrkdkzfxg2zm5jvadx970s2ggfi9kzni-fusuma-1.3.0/bin/.fusuma-wrapped:20:in `<main>'
	25: from /nix/store/zrkdkzfxg2zm5jvadx970s2ggfi9kzni-fusuma-1.3.0/bin/.fusuma-wrapped:20:in `load'
	24: from /nix/store/zrkdkzfxg2zm5jvadx970s2ggfi9kzni-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/exe/fusuma:41:in `<top (required)>'
	23: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma.rb:18:in `run'
	22: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma.rb:89:in `run'
	21: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:21:in `run'
	20: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:56:in `debug_events'
	19: from /nix/store/yyhykbfxj6jfc62x4f9a2awfbch605fd-ruby-2.6.6/lib/ruby/2.6.0/open3.rb:101:in `popen3'
	18: from /nix/store/yyhykbfxj6jfc62x4f9a2awfbch605fd-ruby-2.6.6/lib/ruby/2.6.0/open3.rb:219:in `popen_run'
	17: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:57:in `block in debug_events'
	16: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:57:in `each'
	15: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:58:in `block (2 levels) in debug_events'
	14: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:22:in `block in run'
	13: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma.rb:90:in `block in run'
	12: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma.rb:100:in `filter'
	11: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma.rb:100:in `any?'
	10: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma.rb:100:in `block in filter'
	 9: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/plugin/filters/filter.rb:17:in `filter'
	 8: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/plugin/filters/libinput_device_filter.rb:29:in `keep?'
	 7: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/plugin/filters/libinput_device_filter.rb:54:in `all'
	 6: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:39:in `available'
	 5: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:33:in `all'
	 4: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:62:in `fetch_devices'
	 3: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:81:in `generate_devices'
	 2: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:81:in `each_with_object'
	 1: from /nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:81:in `each'
/nix/store/75lfi07hzgsiphr8rqss10zhvz213a1w-ruby2.6.6-fusuma-1.3.0/lib/ruby/gems/2.6.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:90:in `block in generate_devices': undefined method `assign_attributes' for nil:NilClass (NoMethodError)
``` 

**To Reproduce**
Steps to reproduce the behavior:
1. Get a Wacom CTH-480 (I suspect that other Wacom tablets might work as well)
2. Start Fusuma with tablet connected
3. Touch your tablet

**Expected behavior**
Recognized gestures from tablet

**Versions**
1.3.0, on NixOS

**Additional context**
Default config)

iberianpig, 2021-03-30:
Please paste the output of libinput-list-devices of Wacom pen tablets. This is very useful for debugging.

cab404, 2021-03-30:
@iberianpig 
```
Device:           Wacom Intuos PT S Pen
Kernel:           /dev/input/event20
Group:            5
Seat:             seat0, default
Size:             152x95mm
Capabilities:     tablet
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   none
Rotation:         n/a

Device:           Wacom Intuos PT S Finger
Kernel:           /dev/input/event21
Group:            5
Seat:             seat0, default
Size:             152x95mm
Capabilities:     pointer gesture
Tap-to-click:     enabled
Tap-and-drag:     enabled
Tap drag lock:    disabled
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   *two-finger edge
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a
```

cab404, 2021-06-12:
@iberianpig If you need any additional info, feel free to call me in Matrix (cab404) or Telegram (CabiaRangris))

iberianpig, 2021-06-12:
@cab404 Did you try to install fusuma from Rubygems with gem command?

I noticed you might be using another package manager.

> Versions
> 1.3.0, on NixOS

This version is so old; you should update to the latest version of fusuma.