iberianpig/fusuma #228: Uninstall Fusuma

huyhoang8398, 2021-04-20:
**Describe the bug**
Cannot uninstall fusuma

**To Reproduce**
Steps to reproduce the behavior:
1. sudo gem install fusuma
2. sudo gem uninstall fusuma
3. Removing fusuma
ERROR:  While executing gem ... (NameError)
    uninitialized constant Gem::RDoc

**Expected behavior**

Uninstall fusuma successful

**Versions**
You can get this information from copy and paste the output of `fusuma --version` from the command line. Also, please include the OS and what version of the OS you're running.
OS: Archlinux

catalyst17, 2021-04-22:
Yo, had the same problem. Try `sudo gem install rdoc` first :)

iberianpig, 2021-04-22:
Is this an Archlinux related issue?

catalyst17, 2021-04-22:
> Is this an Archlinux related issue?

Seems to be it, yes.By the way, I started to have another issue now, with fusuma stopping working after a while, no matter,

iberianpig, 2021-04-22:
> Seems to be it, yes.

Thank you very much. I don't know the details, but it may be that the Ruby installed from pacman does not have rdoc.

---

>By the way, I started to have another issue now, with fusuma stopping working after a while, no matter, has it been started on autostart or by the fusuma -d command. Do I need to create the separate issue? :(

We apologize for the inconvenience, but please submit another issue.

As a hint to find out why a running script stops, it would be helpful to have the logs.
Please try `$ fusuma > logs.txt ` to see if it reproduces.

huyhoang8398, 2021-04-24:
> By the way, I started to have another issue now, with fusuma stopping working after a while, no matter, has it been started on autostart or by the `fusuma -d` command. Do I need to create the separate issue? :(

I think i will close this issue, can you open a new case for your problem @catalyst17.
Thanks @catalyst17 @iberianpig for your answers

closed 2021-04-24 by huyhoang8398