iberianpig/fusuma #229: Fusuma not accepting any gestures

linusrg1, 2021-04-25:
I decided to use the default config from the source code notes as it seemed to have everything I want, but fusuma doesnt want to accept any of my gestures. 

Setup:
KDE Neon
Kernel 5.4
Synaptics trackpad from 11 years ago

iberianpig, 2021-04-25:
Hi @linusrg1 
Please paste your `config.yml` and paste the output of `$ fusuma --version`.

linusrg1, 2021-04-25:
```swipe:
  3:
    left:
      command: "xdotool key alt+Right" # History forward
    right:
      command: "xdotool key alt+Left" # History back
    up:
      command: "xdotool key super" # Activity
    down:
      command: "xdotool key super" # Activity
  4:
    left:
      command: "xdotool key ctrl+alt+Down" # Switch to next workspace
    right:
      command: "xdotool key ctrl+alt+Up" # Switch to previous workspace
    up:
      command: "xdotool key ctrl+alt+Down" # Switch to next workspace
    down:
      command: "xdotool key ctrl+alt+Up" # Switch to previous workspace
pinch:
  in:
    command: "xdotool keydown ctrl click 4 keyup ctrl" # Zoom in
  out:
    command: "xdotool keydown ctrl click 5 keyup ctrl" # Zoom out
```
```I, [2021-04-25T10:15:00.487591 #3017]  INFO -- : reload config: /home/linus/.config/fusuma/config.yml
I, [2021-04-25T10:15:00.600028 #3017]  INFO -- : ---------------------------------------------
I, [2021-04-25T10:15:00.600531 #3017]  INFO -- : Fusuma: 2.0.4
I, [2021-04-25T10:15:00.606141 #3017]  INFO -- : libinput: 1.15.5
I, [2021-04-25T10:15:00.606219 #3017]  INFO -- : ruby 2.7.0p0
I, [2021-04-25T10:15:00.607789 #3017]  INFO -- : OS: Linux 5.11.0-051100-generic #202102142330 SMP Sun Feb 14 23:33:21 UTC 2021
I, [2021-04-25T10:15:00.609775 #3017]  INFO -- : Distribution: KDE neon 5.21 \n \l
I, [2021-04-25T10:15:00.611418 #3017]  INFO -- : Desktop session: plasma x11
I, [2021-04-25T10:15:00.611488 #3017]  INFO -- : ---------------------------------------------
I, [2021-04-25T10:15:00.611538 #3017]  INFO -- : Enabled Plugins: 
I, [2021-04-25T10:15:00.611675 #3017]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-04-25T10:15:00.611729 #3017]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2021-04-25T10:15:00.611775 #3017]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-04-25T10:15:00.611819 #3017]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-04-25T10:15:00.611862 #3017]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-04-25T10:15:00.611906 #3017]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2021-04-25T10:15:00.611950 #3017]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-04-25T10:15:00.612034 #3017]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-04-25T10:15:00.612081 #3017]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-04-25T10:15:00.612121 #3017]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-04-25T10:15:00.612163 #3017]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-04-25T10:15:00.612205 #3017]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2021-04-25T10:15:00.612246 #3017]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2021-04-25T10:15:00.612289 #3017]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2021-04-25T10:15:00.612331 #3017]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2021-04-25T10:15:00.612372 #3017]  INFO -- : ---------------------------------------------
```
here ya go

iberianpig, 2021-04-25:
Did you paste it correctly? There seems to be no swipe in config.

linusrg1, 2021-04-25:
Just accedently left it out in paste to github. It is in the config file on my pc though

iberianpig, 2021-04-25:
Please paste correctly from the start line of the command.

Try swiping with three fingers after running fusuma in your terminal. 
If you get some new output from it, it means that the gesture is recognized.

linusrg1, 2021-04-25:
I tried that I dont get any new output though.

iberianpig, 2021-04-26:
If there is no output, if you set it according to the README and nothing is printed, it means that libinput is not supported.

closed 2021-04-26 by iberianpig