iberianpig/fusuma #230: Assigning keys for 3 finger tap in fusuma config.yml file

gaming-wiz, 2021-04-25:
I managed to copy paste the below syntax into config.yml file, but I am not sure about how to assign keys since I am new to Ubuntu.
Now for example, how should I should assign 3 finger tap for super+a key?

tap:
  1:
    command: "echo ----------------tap1----------------------------"
  2:
    command: "echo ----------------tap2----------------------------"
  3:
    command: "echo ----------------tap3----------------------------"
  4:
    command: "echo ----------------tap4----------------------------"

hold:
  1:
    command: "echo ----------------hold1----------------------------"
  2:
    command: "echo ----------------hold2----------------------------"
  3:
    command: "echo ----------------hold3----------------------------"
  4:
    command: "echo ----------------hold4----------------------------"

plugin: 
  inputs:
    libinput_command_input:
      enable-tap: true
      verbose: true

iberianpig, 2021-04-25:
Hi @gaming-wiz 

Set the value of the command: property as a command that can be executed in the terminal.
 Your settings are same with example, it just prints the text on the terminal.

 If you have xdotool installed, you can immediately try it in that terminal.

0. Check man page of xdotool
1. Try xdotool command on your terminal
2. Check it works correctly
3. Paste the command you ran into config.yml

gaming-wiz, 2021-04-25:
Thanks got it, I realized that the syntax was same as that used for swipe feature of fusuma.
For any one who's struggling with this, you can see that I assigned super+d for 3 finger tap in config.yml file. The same syntax holds good for other tap/hold features.

tap:
  1:
    command: "echo ----------------tap1----------------------------"
  2:
    command: "echo ----------------tap2----------------------------"
  3:
    command: "xdotool key super+d"
  4:
    command: "echo ----------------tap4----------------------------"

hold:
  1:
    command: "echo ----------------hold1----------------------------"
  2:
    command: "echo ----------------hold2----------------------------"
  3:
    command: "echo ----------------hold3----------------------------"
  4:
    command: "echo ----------------hold4----------------------------"

plugin: 
  inputs:
    libinput_command_input:
      enable-tap: true
      verbose: true

gaming-wiz, 2021-04-25:
>    1. Check man page of xdotool
>    2. Try xdotool command on your terminal
>    3. Check it works correctly
>    4. Paste the command you ran into config.yml

Thanks, I got it.
Also, I wanted to know if two finger swipe feature is available in fusuma.
If not, please try to include it in the upcoming updates or suggest some
alternatives for this.

closed 2021-04-25 by gaming-wiz

iberianpig, 2021-04-25:
Currently I have no plans to implement a two-finger swipe. This is because libinput itself is designed to be treated as just a two-finger scroll.
Of course, it's not impossible. However, since the native two-finger scroll cannot be dynamically disabled, scrolling and swipe action will trigger at the same time. So I don't think it's a useful gesture.