iberianpig/fusuma #232: Various issues found in v2 - 4 fingers swipe, pinch, drag

cani-venatici, 2021-05-01:
#### Describe the bug
After updating fusuma to v2 I found the following issues which were not present in the previous version:

1. RightAlt+4-Fingers-Swipe in any direction stopped working while 4Fingers swipe and any combination of it with ctrl or shift is unaffected. Libinput recognizes the keypress and the 4-swipe. It was and is currently bound to

```yml
  4:
    left:
      sendkey: "LEFTMETA+LEFT" # Move windows to left size of current monitor
      threshold: 0.3
      interval: 0.7
      keypress:
        RIGHTSHIFT:
          sendkey: "LEFTCTRL+LEFTALT+LEFTSHIFT+LEFT" # Move window to previous workspace
        RIGHTALT:
          sendkey: "LEFTSHIFT+LEFTMETA+LEFT" # Move windows to left monitor - stopped working after v2
        RIGHTCTRL:
          sendkey: "LEFTCTRL+LEFTALT+RIGHT" # Switch to next workspace + visual feedback
```

2. Three fingers drag activates with a delay and doesn't drag content (windows).

3. can't select submenu entries in gnome topbar, for example I can expand the poweroff/restart menu but selecting the poweroff entry has no effect and the submenu quits.

Could the tap entry in config be the culprit? Selecting entries in gnome topbar menus worked prior to fusuma v2 with the 1 tap command commented out, though.

```yml
tap:
#    1:
#      command: ""
```

4. 4 fingers pinch are inverted, in works as out and viceversa. The command assigned to 4-pinch **out** gets executed on 4-pinch **in** and viceversa.

5. Tap and Hold don't work for libinput > v16.4

#### Versions
```shell
fusuma --version
I, [2021-05-01T20:02:35.527623 #44867]  INFO -- : reload config: /home/user/.config/fusuma/config.yml
I, [2021-05-01T20:02:35.608508 #44867]  INFO -- : ---------------------------------------------
I, [2021-05-01T20:02:35.608791 #44867]  INFO -- : Fusuma: 2.0.4
I, [2021-05-01T20:02:35.609311 #44867]  INFO -- : libinput: 1.16.4
I, [2021-05-01T20:02:35.609335 #44867]  INFO -- : ruby 3.0.1p64
I, [2021-05-01T20:02:35.609754 #44867]  INFO -- : OS: Linux 5.11.16-2-MANJARO #1 SMP PREEMPT Thu Apr 22 09:53:11 UTC 2021
I, [2021-05-01T20:02:35.610176 #44867]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2021-05-01T20:02:35.613272 #44867]  INFO -- : Desktop session: gnome x11
I, [2021-05-01T20:02:35.613295 #44867]  INFO -- : ---------------------------------------------
I, [2021-05-01T20:02:35.613314 #44867]  INFO -- : Enabled Plugins: 
I, [2021-05-01T20:02:35.613386 #44867]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-05-01T20:02:35.613402 #44867]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2021-05-01T20:02:35.613415 #44867]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2021-05-01T20:02:35.613425 #44867]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2021-05-01T20:02:35.613436 #44867]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2021-05-01T20:02:35.613446 #44867]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-05-01T20:02:35.613456 #44867]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-05-01T20:02:35.613466 #44867]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-05-01T20:02:35.613476 #44867]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2021-05-01T20:02:35.613485 #44867]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2021-05-01T20:02:35.613496 #44867]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-05-01T20:02:35.613506 #44867]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-05-01T20:02:35.613516 #44867]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2021-05-01T20:02:35.613629 #44867]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-05-01T20:02:35.613642 #44867]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-05-01T20:02:35.613652 #44867]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2021-05-01T20:02:35.613661 #44867]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2021-05-01T20:02:35.613671 #44867]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2021-05-01T20:02:35.613680 #44867]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-05-01T20:02:35.613690 #44867]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2021-05-01T20:02:35.613699 #44867]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2021-05-01T20:02:35.613708 #44867]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2021-05-01T20:02:35.613718 #44867]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2021-05-01T20:02:35.613727 #44867]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2021-05-01T20:02:35.613736 #44867]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2021-05-01T20:02:35.613746 #44867]  INFO -- : ---------------------------------------------
```

#### Configuration
```yml
## Swipes

swipe:
  3:
  # Three finger drag
    begin:
      command: xdotool mousedown 1
    update:
      command: xdotool mousemove_relative -- $move_x, $move_y
      interval: 0.01
      accel: 2
    end:
      command: xdotool mouseup 1
   ##
    left: 
      sendkey: "LEFTALT+RIGHT"
      threshold: .8
      interval: 2.5

    right: 
      sendkey: "LEFTALT+LEFT"
      threshold: .8
      interval: 2.5

    up:
      sendkey: "LEFTCTRL+PAGEDOWN"
      threshold: .3
      interval: .7
      keypress:
        RIGHTCTRL:
          sendkey: "LEFTCTRL+t"

    down:
      sendkey: "LEFTCTRL+PAGEUP"
      threshold: .3
      interval: .7
      keypress:
        RIGHTALT:
          sendkey: "LEFTCTRL+w"

  4:
    left:
      sendkey: "LEFTMETA+LEFT" # Move windows to left size of current monitor
      threshold: 0.3
      interval: 0.7
      keypress:
        RIGHTSHIFT:
          sendkey: "LEFTCTRL+LEFTALT+LEFTSHIFT+LEFT" # Move window to previous workspace
        RIGHTALT:
          sendkey: "LEFTSHIFT+LEFTMETA+LEFT" # Move windows to left monitor
        RIGHTCTRL:
          sendkey: "LEFTCTRL+LEFTALT+RIGHT" # Switch to next workspace + visual feedback

    right:
      sendkey: "LEFTMETA+RIGHT" # Move windows to right size of current monitor
      threshold: 0.3
      interval: 0.7
      keypress:
        RIGHTSHIFT: 
          sendkey: "LEFTCTRL+LEFTALT+LEFTSHIFT+RIGHT" # Move window to next workspace
        RIGHTALT:
          sendkey: "LEFTSHIFT+LEFTMETA+RIGHT" # Move windows to right monitor
        RIGHTCTRL:
          sendkey: "LEFTCTRL+LEFTALT+LEFT" # Switch to previous workspace + visual feedback

    up: 
      window:
        maximized: toggle # Make window full size

    down:
      command: "dbus-send --session --type=method_call --dest=org.gnome.Shell /org/gnome/Shell org.gnome.Shell.Eval string:'Main.overview.toggle();'" # Activities overview
      threshold: 1.2
      interval: 0.7
      keypress:
        RIGHTALT:
          sendkey: "LEFTALT+F4" # Close focused window
        RIGHTCTRL:
          sendkey: "LEFTMETA+H" # Minimizes window

## Pinches

pinch:
    2:
        in:
            command: ""
            threshold: 9
            interval: 12
        out:
            command: ""
            threshold: 9
            interval: 12
    3:
        in:
            sendkey: VOLUMEUP #test
#            command: "xdotool key LEFTCTRL+LEFTALT+XF86AudioRaiseVolume" # Volume up
            threshold: 1
            interval: 0.4
        out:
            sendkey: VOLUMEDOWN #test
#            command: "xdotool key LEFTCTRL+LEFTALT+XF86AudioLowerVolume" # Volume down
            threshold: 1
            interval: 0.4

    4:
    # For some reason 'in' works as out and 'out' works as in
        in:
            sendkey: "LEFTCTRL+V" # Copy
            threshold: .1
            interval: 3
        out:
            sendkey: "LEFTCTRL+C" # Paste
            threshold: .1
            interval: 3

## Rotation volume
rotate:
  3:
    clockwise:
      update:
        sendkey: VOLUMEUP
    counterclockwise:
      update:
        sendkey: VOLUMEDOWN

## Taps and Holds - tapping/holding don't work for libinput > 16.4
tap:
#    1:
#      command: ""
    2:
      command: "xdotool click 3"
    3:
      command: "xdotool click 2"
    4:
      window: "fullscreen"

hold:
    1:
      window: ""
    2:
      sendkey: "ENTER"
    3:
      sendkey: "LEFTCTRL+W" # Close
#    4:
#      sendkey: "LEFTALT+F7" # Move window
#      keypress:
#        RIGHTCTRL:
#          sendkey: "LEFTALT+F8" # Resize window

## Plugin specs

plugin: 
  inputs:
    libinput_command_input:
      enable-tap: true # Click to tap
      enable-dwt: true # Disable tap while typing
      show-keycodes: true
      verbose: true
  executors:
    wmctrl_executor:
      wrap-navigation: true
```

iberianpig, 2021-05-01:
> 4 fingers pinch are inverted, in works as out and viceversa. The command assigned to 4-pinch out gets executed on 4-pinch in and viceversa.

Pinch In/Out has been backward for quite some time。

I've only just noticed this now. Thanks for that.

I had mistakenly thought that my Pinch In and Zoom In were the same😅

iberianpig, 2021-05-02:
> RightAlt+4-Fingers-Swipe in any direction stopped working while 4Fingers swipe and any combination of it with ctrl or shift is unaffected.

Are you using any keyremapper like xkeysnail?
I reproduced the problem with xkeysnail and fusuma v2.
After stopping xkeysnail, it works without the problem.

cani-venatici, 2021-05-02:
> Are you using any keyremapper like xkeysnail?
> I reproduced the problem with xkeysnail and fusuma v2.
> After stopping xkeysnail, it works without the problem.

[image]

I'm not using any key remappers but I guess gnome 38 is acting as one. I updated to gnome 40 and noticed the 'alternate characters key' option in keyboard settings. I don't think that option exists in gnome 38 keyboard settings. As far as I recall only the 'compose key' option existed but it was in gnome-tweaks (manjaro). Maybe a gnome update introduced that setting and messed up things? Anyway changing the 'alternate character key' to any other key not used by fusuma for gestures enables fusuma to recognize them.

However, now the RightAlt modifier is included in the keystrokes that get executed.

```yml
# current cfg
  4:
    left:
      sendkey: "LEFTMETA+LEFT" # Move windows to left size of current monitor
      keypress:
        RIGHTALT:
          sendkey: "LEFTSHIFT+LEFTMETA+LEFT" # Move windows to left monitor
```

When using the 4 finger gesture with the `RIGHTALT` switch, the switch is included in the keystrokes. So instead of `SHIFT+SUPER+LEFT/RIGHT` what gets executed is `ALT+SHIFT+SUPER+LEFT/RIGHT` (Basically the window gets moved to leftward/rightward workspace instead of leftward/rightward monitor).

Calling that shortcut through the keyboard or in a terminal `fusuma-sendkey "LEFTSHIFT+LEFTMETA+LEFT"` works as expected, that is the window is moved to leftward monitor, so I'd exclude conflicting keybindings in gnome.

Changing the command in fusuma config to
```yml
     keypress:
        RIGHTALT:
#          sendkey: "LEFTSHIFT+LEFTMETA+LEFT" # Move windows to left monitor
          command: "xdotool key --clearmodifiers Shift+Super+Left"
```
works as expected.

It seems it's only for the `RIGHTALT + 4-fingers left/right` that the keyswitch is included in the keystroke since any other 3- or 4-fingers gesture combined with `RIGHTALT` work correctly.

Is this madness?

---

> 3. can't select submenu entries in gnome topbar, for example […]

This is no longer an issue in gnome 40, apparently.

iberianpig, 2021-05-04:
>  Maybe a gnome update introduced that setting and messed up things? Anyway changing the 'alternate character key' to any other key not used by fusuma for gestures enables fusuma to recognize them.

I suspect that is the cause.
When one modifier key is released, xkeysnail forces the other modifier keys to be released as well.
So that fusuma doesn't recognize the first modifier key after executed sendkey(which presses and releases some modifier keys)

iberianpig, 2021-05-04:
> However, now the RightAlt modifier is included in the keystrokes that get executed.

That is the behavior I intended.

I changed the sendkey's behavior that when keypress + sendkey is used.  
Since version v0.6.0.pre, fusuma-plugin-sendkey wouldn't release the modifier key used in keypress.

This was a spec change in sendkey, but it may not have been a good idea.

I'll try to think more about how to make it work.

iberianpig, 2021-05-04:
> Tap and Hold don't work for libinput > v16.4

see: https://github.com/iberianpig/fusuma-plugin-tap/issues/13#issuecomment-803597089

iberianpig, 2021-05-05:
@cani-venatici  I released fusuma-plugin-sendkey v0.6.2 

Please check that it fixed https://github.com/iberianpig/fusuma-plugin-sendkey/pull/21

cani-venatici, 2021-05-06:
> @cani-venatici I released fusuma-plugin-sendkey v0.6.2
> 
> Please check that it fixed [iberianpig/fusuma-plugin-sendkey#21](https://github.com/iberianpig/fusuma-plugin-sendkey/pull/21)

Alt + 4-swipe is working as expected on sendkey 0.6.2. Thank you.

iberianpig, 2021-05-07:
> Three fingers drag activates with a delay and doesn't drag content (windows).

I would like to know why fusuma can't Drag the content and how fusuma behaves.
I have confirmed that it is working fine in my environment.

Could you please take a video of the Drag & Drop with Fusuma?

cani-venatici, 2021-05-07:
https://user-images.githubusercontent.com/30556695/117390438-7ac34880-aeee-11eb-8121-a026d85d0ac0.mp4

Tried to move some windows. Hope it's visible despite compression.
When I keep dragging I can feel the acceleration working but the windows don't move. Also when I try to move gedit's window by the titlebar what actually gets moved is the text in the editor 👀.

iberianpig, 2021-05-12:
@cani-venatici I've noticed that keyboard events are firing during swipe events.

[image]

Could you please comment out the 3 fingers' left/right swipe and then check if it works?

cani-venatici, 2021-05-22:
I tried that before posting here. Tried again now with all 3 fingers gestures commented out and it still won't work. It grabs the window for a second and then drops it. Apparently libinput debug doesn't give any feedback on the release. Maybe I should point out i'm using a wacom tablet. This tablet can do 3 fingers drag on win/mac but since it can't on linux I'll just assume it's libinput's fault messing up drags with swipes. I'll keep dragging with double taps. Ty for support.

```shell
xsetwacom lists devices

Wacom Intuos Pro M (WL) Pen stylus	id: 12	type: STYLUS    
Wacom Intuos Pro M (WL) Pen eraser	id: 13	type: ERASER    
Wacom Intuos Pro M (WL) Pen cursor	id: 14	type: CURSOR    
Wacom Intuos Pro M (WL) Pad pad 	id: 15	type: PAD       
Wacom Intuos Pro M (WL) Finger touch	id: 16	type: TOUCH   
```

HritwikSinghal, 2021-07-24:
> > Tap and Hold don't work for libinput > v16.4
> 
> see: [iberianpig/fusuma-plugin-tap#13 (comment)](https://github.com/iberianpig/fusuma-plugin-tap/issues/13#issuecomment-803597089)

hey, have you looked at this MR https://gitlab.freedesktop.org/libinput/libinput/-/merge_requests/602 .

closed 2022-04-28 by stale[bot]