iberianpig/fusuma #233: [question] Yaml interpolation.

NikhilSaini38, 2021-05-05:
I am looking to create configurable yaml configs. do you interpret yaml from scratch or do you use a third party library? is there any way to interpolate/concatenate strings from keys in the yaml?

iberianpig, 2021-05-06:
Hi @NikhilSaini38 

I may not have understood your question properly.

> do you interpret yaml from scratch or do you use a third party library? is there any way to interpolate/concatenate strings from keys in the yaml?

I'm using only the standard ruby library for loading and parsing config.yml.
https://github.com/iberianpig/fusuma/blob/134b784ca6f674c8a9c3844ccf6da0d64346d5c2/lib/fusuma/config/searcher.rb#L12-L32
If it were easy, I could just use only Hash#dig, but since the design has become more difficult, the code has become more complex.

> is there any way to interpolate/concatenate strings from keys in the yaml?

What does this mean? (If you have a specific example of what you want to do, I may be able to answer)

NikhilSaini38, 2021-05-06:
I was attempting to make config files that are user friendly for both left-handed and right-handed users. I thought if there was some way I could just define handedness using , say,
```
keyhand: LEFT
````
then I could define the rest of the configs with keypresses eg, ```LEFTMETA ``` would become say ```{{keyhand}}META```

In the hindsight, it looks stupid quest to me though.

iberianpig, 2021-05-07:
Thank you. This helps me to understand clearly :smile: 

Fusuma does not do anything more than the standard YAML. Therefore, users need to generate config.yml with their own custom programs.

For example, how about using Ruby's standard ERB to generate config.yml from a template?

`variables + config.yml.erb ==compile==> config.yml`

1. Create an ERB template with embedded environment variables as a file with a name like config.yml.erb.
2. Compile it to config.yml before running Fusuma.

closed 2021-05-07 by NikhilSaini38