iberianpig/fusuma #234: Fusuma suddenly not working. Ruby 3 related?

mike-lloyd03, 2021-05-09:
**Describe the bug**
Fusuma starts, reads input, but does not execute any commands. `libinput debug-events` registers all motions (3 and 4 finger swipe and pinch). I haven't changed anything recently except upgrading to Ruby 3.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade to Ruby 3 (I'm not sure if this is the problem but this is the only thing that has changed recently.)
2. Start fusuma at the terminal with the `fusuma` command.
3. Version information and plugin list is displayed but no commands are registered or executed.

**Versions**
```yml
I, [2021-05-08T22:06:37.586080 #619821]  INFO -- : reload config: /home/mike/.config/fusuma/config.yml
I, [2021-05-08T22:06:37.626459 #619821]  INFO -- : ---------------------------------------------
I, [2021-05-08T22:06:37.626696 #619821]  INFO -- : Fusuma: 2.0.4
I, [2021-05-08T22:06:37.628076 #619821]  INFO -- : libinput: 1.17.2
I, [2021-05-08T22:06:37.628099 #619821]  INFO -- : ruby 3.0.1p64
I, [2021-05-08T22:06:37.628591 #619821]  INFO -- : OS: Linux 5.12.0-1-MANJARO #1 SMP PREEMPT Mon Apr 26 22:09:28 UTC 2021
I, [2021-05-08T22:06:37.629048 #619821]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2021-05-08T22:06:37.631237 #619821]  INFO -- : Desktop session: plasma x11
I, [2021-05-08T22:06:37.631259 #619821]  INFO -- : ---------------------------------------------
I, [2021-05-08T22:06:37.631269 #619821]  INFO -- : Enabled Plugins: 
I, [2021-05-08T22:06:37.631370 #619821]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-05-08T22:06:37.631404 #619821]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2021-05-08T22:06:37.631435 #619821]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-05-08T22:06:37.631445 #619821]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-05-08T22:06:37.631453 #619821]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-05-08T22:06:37.631462 #619821]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2021-05-08T22:06:37.631470 #619821]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-05-08T22:06:37.631478 #619821]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-05-08T22:06:37.631486 #619821]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-05-08T22:06:37.631517 #619821]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-05-08T22:06:37.631547 #619821]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-05-08T22:06:37.631554 #619821]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2021-05-08T22:06:37.631563 #619821]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2021-05-08T22:06:37.631572 #619821]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2021-05-08T22:06:37.631580 #619821]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2021-05-08T22:06:37.631602 #619821]  INFO -- : ---------------------------------------------
```

**Configuration**
```yml
plugin:
  executors:
    sendkey_executor:
      device_name: 'Razer Razer Blade'
swipe:
  3: 
    left: 
      sendkey: 'LEFTMETA+H'
    right: 
      sendkey: 'LEFTMETA+L'
    up:
      sendkey: 'LEFTMETA+K'
  4:
    left:
      sendkey: 'LEFTMETA+LEFT'
    right:
      sendkey: 'LEFTMETA+RIGHT'
pinch:
  in:
    sendkey: 'LEFTALT+EQUAL'
  out:
    sendkey: 'LEFTALT+MINUS'

threshold:
  swipe: 0.5
  pinch: 0.125

interval:
  swipe: 0.5
  pinch: 0.5
```

iberianpig, 2021-05-09:
Hi, @mike-lloyd03 

I think you haven't installed fusuma-plugin-sendkey.

see: https://github.com/iberianpig/fusuma-plugin-sendkey#installation

```sh
I, [2021-05-08T22:06:37.631269 #619821]  INFO -- : Enabled Plugins: 
I, [2021-05-08T22:06:37.631370 #619821]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-05-08T22:06:37.631404 #619821]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2021-05-08T22:06:37.631435 #619821]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-05-08T22:06:37.631445 #619821]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-05-08T22:06:37.631453 #619821]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-05-08T22:06:37.631462 #619821]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2021-05-08T22:06:37.631470 #619821]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-05-08T22:06:37.631478 #619821]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-05-08T22:06:37.631486 #619821]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-05-08T22:06:37.631517 #619821]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-05-08T22:06:37.631547 #619821]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-05-08T22:06:37.631554 #619821]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2021-05-08T22:06:37.631563 #619821]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2021-05-08T22:06:37.631572 #619821]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2021-05-08T22:06:37.631580 #619821]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2021-05-08T22:06:37.631602 #619821]  INFO -- : ---------------------------------------------
```

mike-lloyd03, 2021-05-10:
Aye aye aye what a simple fix. That's the problem. Thanks for getting back to me. Got it running now after installing `fusuma-plugin-sendkey`

closed 2021-05-10 by mike-lloyd03