iberianpig/fusuma #235: Possibility to use less than 3 fingers to make gestures

iacchus, 2021-05-13:
**Is your feature request related to a problem? Please describe.**

Not sure.

**Describe the solution you'd like**

I would like to be able to use swipe with one or two fingers.

**Describe alternatives you've considered**

Lowering the number of fingers needed to make the gestures.

**Additional context**

I use a mouse and would like to use the touchpad to change workspaces swiping left / right with only one finger. 

I would like to know if it is possible to use 1 or 2 fingers to swipe etc. Thank you for the software!

iberianpig, 2021-05-24:
Currently, Fusuma does not support one- and two-finger gestures.

Even if you get 1 and 2 finger gesture settings, you won't be useful unless they hijack the normal 1 finger cursor movement or tap 2 finger scrolling.
Because Fusuma does not hijack the default behavior on the touchpad. Fusuma just parses touchpad logs and triggers commands additionally.

codeyash, 2021-06-02:
Apple trackpad does support 2 finger swipe. I always in my Mac OS. Trying to use the same trackpad with Arch Linux. Using libinput and fusuma I able to achieve many behaviors except 2 finger swipe. It can be used like alt+Left or alt_Right which can be used anywhere like browsers back and forward. The trackpad does support it but I don't know how to use it using fusuma.

Any help or instruction is appreciated.

codeyash, 2021-06-02:
xinput list-props "Apple Inc. Magic Trackpad 2"
Device 'Apple Inc. Magic Trackpad 2':
	Device Enabled (161):	1
	Coordinate Transformation Matrix (163):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Tapping Enabled (316):	1
	libinput Tapping Enabled Default (317):	0
	libinput Tapping Drag Enabled (318):	1
	libinput Tapping Drag Enabled Default (319):	1
	libinput Tapping Drag Lock Enabled (320):	0
	libinput Tapping Drag Lock Enabled Default (321):	0
	libinput Tapping Button Mapping Enabled (322):	0, 1
	libinput Tapping Button Mapping Default (323):	1, 0
	libinput Natural Scrolling Enabled (296):	1
	libinput Natural Scrolling Enabled Default (297):	0
	libinput Scroll Methods Available (298):	1, 1, 0
	libinput Scroll Method Enabled (299):	1, 0, 0
	libinput Scroll Method Enabled Default (300):	1, 0, 0
	libinput Click Methods Available (324):	1, 1
	libinput Click Method Enabled (325):	0, 1
	libinput Click Method Enabled Default (326):	0, 1
	libinput Middle Emulation Enabled (305):	0
	libinput Middle Emulation Enabled Default (306):	0
	libinput Accel Speed (307):	1.000000
	libinput Accel Speed Default (308):	0.000000
	libinput Accel Profiles Available (309):	1, 1
	libinput Accel Profile Enabled (310):	1, 0
	libinput Accel Profile Enabled Default (311):	1, 0
	libinput Left Handed Enabled (312):	0
	libinput Left Handed Enabled Default (313):	0
	libinput Send Events Modes Available (281):	1, 0
	libinput Send Events Mode Enabled (282):	0, 0
	libinput Send Events Mode Enabled Default (283):	0, 0
	Device Node (284):	"/dev/input/event16"
	Device Product ID (285):	1452, 613
	libinput Drag Lock Buttons (314):	<no items>
	libinput Horizontal Scroll Enabled (315):	1

closed 2022-04-28 by stale[bot]