iberianpig/fusuma #236: Touch input detected but ignored?

lucianoiam, 2021-05-18:
This software looks promising!

I have a monitor whose touchscreen is detected as "WingCoolTouch" by libinput. It is verified to pass this touch capability test on Firefox/Linux: http://dev.openlayers.org/examples/multitouch.html

Running `fusuma -v` shows it is correctly detected. Note there are two identically named input devices, one for the touch device itself and the other emulating a mouse, probably for compatibility with systems that cannot deal with a touchscreen:

```
D, [2021-05-18T23:47:30.547025 #6940] DEBUG -- : {:input_event=>time: 2021-05-18 23:47:30 +0200, tag: libinput_command_input, record: -event6   DEVICE_ADDED     WingCoolTouch WingCoolTouch       seat0 default group4  cap:t  size 216x91mm ntouches 10 calib}
D, [2021-05-18T23:47:30.547280 #6940] DEBUG -- : {:input_event=>time: 2021-05-18 23:47:30 +0200, tag: libinput_command_input, record: -event8   DEVICE_ADDED     WingCoolTouch WingCoolTouch       seat0 default group4  cap:p  size 216x91mm left scroll-nat calib}
```

Later on the log, the first "WingCoolTouch WingCoolTouch" (the touch variant, with id `event6`) does not appear listed as an available device:

```
D, [2021-05-18T23:47:31.501599 #6940] DEBUG -- : {:available_devices=>[#<Fusuma::Device:0x000055fe55ae9210 @id="event8", @name="WingCoolTouch WingCoolTouch", @capabilities="pointer", @available=true>, #<Fusuma::Device:0x000055fe55a5b320 @id="event4", @name="SEM USB Keyboard Consumer Control", @capabilities="keyboard pointer", @available=true>, #<Fusuma::Device:0x000055fe5599f8f0 @id="event2", @name="Chicony  Wireless Device Mouse", @capabilities="pointer", @available=true>]}
```

Surprisingly or not, verbose output shows activity from `event6` when touching the screen:

```
D, [2021-05-19T00:15:00.437597 #8531] DEBUG -- : {:input_event=>time: 2021-05-19 00:15:00 +0200, tag: libinput_command_input, record:  event6   TOUCH_MOTION     +4.349s	0 (0) 75.22/39.79 (162.16/36.04mm)}
D, [2021-05-19T00:15:00.437836 #8531] DEBUG -- : {:input_event=>time: 2021-05-19 00:15:00 +0200, tag: libinput_command_input, record:  event6   TOUCH_MOTION     +4.349s	1 (1) 68.76/39.27 (148.24/35.57mm)}
D, [2021-05-19T00:15:00.438027 #8531] DEBUG -- : {:input_event=>time: 2021-05-19 00:15:00 +0200, tag: libinput_command_input, record:  event6   TOUCH_MOTION     +4.349s	2 (2) 62.30/50.02 (134.32/45.30mm)}
D, [2021-05-19T00:15:00.438166 #8531] DEBUG -- : {:input_event=>time: 2021-05-19 00:15:00 +0200, tag: libinput_command_input, record:  event6   TOUCH_FRAME      +4.349s	
```

But my configured gesture is never detected, verbose output does not show any hint of that (should it?) and the  command is never triggered:

```
swipe:
  3:
    right:
      command: "xdotool mousemove 1 900"
```

Not sure this unfortunate coincidence of having the same input device name for both touch and pointer means trouble for fusuma but it is worth mentioning.

Here is an excerpt from running `libinput list-devices` on the same box :
 
```
Device:           WingCoolTouch WingCoolTouch
Kernel:           /dev/input/event6
Group:            4
Seat:             seat0, default
Size:             216x91mm
Capabilities:     touch 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           WingCoolTouch WingCoolTouch
Kernel:           /dev/input/event8
Group:            4
Seat:             seat0, default
Size:             216x91mm
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a
```

iberianpig, 2021-05-24:
Unfortunately, Fusuma only recognizes `/GESTURE_(SWIPE|PINCH)_(BEGIN|UPDATE|END)/` for touchpad events.

see: https://github.com/iberianpig/fusuma/blob/master/lib/fusuma/plugin/parsers/libinput_gesture_parser.rb#L40

Currently, Fusuma is made for touchpads, so we need to create a new parser or detector as touchscreen as a fusuma-plugin.

lucianoiam, 2021-05-24:
I see... this is a case of RTFM, sorry. Unfortunately I cannot help here because I'm neither in the Ruby or libinput league. I was exploring how to bring up the XFCE panel using a touchscreen gesture. 

Keep up the great work, thanks.