iberianpig/fusuma #241: Incorrect windowing behavior.

Killllero0, 2021-08-05:
**Describe the bug**

Fusuma on KDE5 (Manjaro Linux) transfers the work table and all work tables break. 

**To Reproduce**
Steps to reproduce the behavior:
1. Hold down right shift.
2. Swipe left(right)

**Expected behavior**
Ignoring the gesture. 

**Versions**
```yml
I, [2021-08-05T09:45:23.389618 #4587]  INFO -- : reload config: /home/yunusove/.config/fusuma/config.yml
I, [2021-08-05T09:45:23.541539 #4587]  INFO -- : ---------------------------------------------
I, [2021-08-05T09:45:23.541885 #4587]  INFO -- : Fusuma: 2.0.5
I, [2021-08-05T09:45:23.544642 #4587]  INFO -- : libinput: 1.18.0
I, [2021-08-05T09:45:23.544687 #4587]  INFO -- : ruby 3.0.2p107
I, [2021-08-05T09:45:23.545701 #4587]  INFO -- : OS: Linux 5.13.5-1-MANJARO #1 SMP PREEMPT Mon Jul 26 07:43:29 UTC 2021
I, [2021-08-05T09:45:23.546258 #4587]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2021-08-05T09:45:23.552053 #4587]  INFO -- : Desktop session: plasma x11
I, [2021-08-05T09:45:23.552148 #4587]  INFO -- : ---------------------------------------------
I, [2021-08-05T09:45:23.552188 #4587]  INFO -- : Enabled Plugins: 
I, [2021-08-05T09:45:23.552403 #4587]  INFO -- :   Fusuma::Plugin::Buffers::AppmatcherBuffer
I, [2021-08-05T09:45:23.552442 #4587]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-08-05T09:45:23.552468 #4587]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2021-08-05T09:45:23.552494 #4587]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2021-08-05T09:45:23.552519 #4587]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2021-08-05T09:45:23.552545 #4587]  INFO -- :   Fusuma::Plugin::Detectors::AppmatcherDetector
I, [2021-08-05T09:45:23.552570 #4587]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2021-08-05T09:45:23.552596 #4587]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-08-05T09:45:23.552622 #4587]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-08-05T09:45:23.552647 #4587]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-08-05T09:45:23.552673 #4587]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2021-08-05T09:45:23.552697 #4587]  INFO -- :   Fusuma::Plugin::Events::Records::AppmatcherRecord
I, [2021-08-05T09:45:23.552723 #4587]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2021-08-05T09:45:23.552748 #4587]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-08-05T09:45:23.552774 #4587]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-08-05T09:45:23.552800 #4587]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2021-08-05T09:45:23.552825 #4587]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-08-05T09:45:23.552850 #4587]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-08-05T09:45:23.552875 #4587]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2021-08-05T09:45:23.552900 #4587]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2021-08-05T09:45:23.552925 #4587]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2021-08-05T09:45:23.552950 #4587]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-08-05T09:45:23.552975 #4587]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2021-08-05T09:45:23.553000 #4587]  INFO -- :   Fusuma::Plugin::Inputs::AppmatcherInput
I, [2021-08-05T09:45:23.553025 #4587]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2021-08-05T09:45:23.553050 #4587]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2021-08-05T09:45:23.553075 #4587]  INFO -- :   Fusuma::Plugin::Parsers::AppmatcherParser
I, [2021-08-05T09:45:23.553099 #4587]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2021-08-05T09:45:23.553124 #4587]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2021-08-05T09:45:23.553149 #4587]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2021-08-05T09:45:23.553175 #4587]  INFO -- : ---------------------------------------------
```

**Configuration**
```yml
swipe:
  3:
    up:
      sendkey: "LEFTCTRL+F9"
      keypress:
        LEFTSHIFT:
          sendkey: "LEFTCTRL+F10"
    down:
      sendkey: "LEFTMETA+D"
    left:
      sendkey: "LEFTMETA+LEFTALT+left"
    right:
      sendkey: "LEFTMETA+LEFTALT+right"

  4:
    up:
      window:
        fullscreen: add
    down:
      window:
        fullscreen: remove
    left:
      workspace: 'next'
      keypress:
        LEFTSHIFT:
          window: 'next'
    right:
      workspace: 'prev'
      keypress:
        LEFTSHIFT:
          window: 'prev'

plugin:
  executors:
    wmctrl_executor:
      wrap-navigation: true
  inputs:
    libinput_command_input:
      show-keycodes: true

```

Killllero0, 2021-12-21:
Sorry for the long wait. Here's a video.
https://youtu.be/ke0J_Fa2e1g

iberianpig, 2026-10-03:
This issue has been open since Fusuma 2.0.5 and has no reproducible details beyond the attached video. It is already marked , and the reported behavior appears to be specific to the KDE/window-management setup rather than an actionable Fusuma defect. Closing for now. If this still affects a current Fusuma release, please open a new issue with the current version, desktop session, configuration, and reproduction steps.

closed 2026-10-03 by iberianpig