iberianpig/fusuma #246: Fusuma doesn't work with Libinput10 v1.19.1-1

jJit0, 2021-10-02:
Fusuma stopped working after upgrading libinput10 to version 1.19.1-1

I've also upgraded fusuma to the latest version this morning, but I think the problem comes from libinput.

Any ideas ?

iberianpig, 2021-10-02:
Thanks for reporting.
I'll update to libinput 1.19.1 tomorrow and check the behavior.

BTW, if you want to fix libinput version, build libinput locally and specify its path in config.yml as shown below. (I often use this configuration method for testing purposes.) 

See: https://github.com/iberianpig/fusuma/wiki/Use-latest-libinput-on-fusuma

The following is my config.yml

```yaml
plugin:
  inputs:
    libinput_command_input:
      libinput-list-devices: /home/iberianpig/.ghq/gitlab.freedesktop.org/libinput/libinput/builddir/1.17.0/libinput-list-devices
      libinput-debug-events: /home/iberianpig/.ghq/gitlab.freedesktop.org/libinput/libinput/builddir/1.17.0/libinput-debug-events

```

jJit0, 2021-10-02:
Arigato iberianpig-san, I'll wait for your update.

I prefer not to use a manual build version of libinput, it can brick my production laptop.
When I have some time, I'll try it on a test purpose laptop.

Regards,

iberianpig, 2021-10-03:
<details>
<summary> It works with Fusuma with libinput: 1.19.1 </summary>

```
I, [2021-10-03T09:20:32.851340 #1164166]  INFO -- : ---------------------------------------------
I, [2021-10-03T09:20:32.851648 #1164166]  INFO -- : Fusuma: 2.1.0
I, [2021-10-03T09:20:32.852841 #1164166]  INFO -- : libinput: 1.19.1
I, [2021-10-03T09:20:32.852883 #1164166]  INFO -- : ruby 3.0.1p64
I, [2021-10-03T09:20:32.853541 #1164166]  INFO -- : OS: Linux 5.10.0-1045-oem #47-Ubuntu SMP Wed Aug 18 10:41:03 UTC 2021
I, [2021-10-03T09:20:32.854191 #1164166]  INFO -- : Distribution: Ubuntu 20.04.3 LTS \n \l
I, [2021-10-03T09:20:32.854750 #1164166]  INFO -- : Desktop session: ubuntu x11
I, [2021-10-03T09:20:32.854765 #1164166]  INFO -- : ---------------------------------------------
I, [2021-10-03T09:20:32.854774 #1164166]  INFO -- : Enabled Plugins:
I, [2021-10-03T09:20:32.854893 #1164166]  INFO -- :   Fusuma::Plugin::Buffers::AppmatcherBuffer
I, [2021-10-03T09:20:32.854912 #1164166]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-10-03T09:20:32.854924 #1164166]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2021-10-03T09:20:32.854937 #1164166]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2021-10-03T09:20:32.854949 #1164166]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2021-10-03T09:20:32.854974 #1164166]  INFO -- :   Fusuma::Plugin::Detectors::AppmatcherDetector
I, [2021-10-03T09:20:32.854989 #1164166]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2021-10-03T09:20:32.855002 #1164166]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-10-03T09:20:32.855025 #1164166]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-10-03T09:20:32.855038 #1164166]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-10-03T09:20:32.855050 #1164166]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2021-10-03T09:20:32.855059 #1164166]  INFO -- :   Fusuma::Plugin::Events::Records::AppmatcherRecord
I, [2021-10-03T09:20:32.856342 #1164166]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2021-10-03T09:20:32.856381 #1164166]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-10-03T09:20:32.856396 #1164166]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-10-03T09:20:32.856414 #1164166]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2021-10-03T09:20:32.856430 #1164166]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-10-03T09:20:32.856445 #1164166]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-10-03T09:20:32.856461 #1164166]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2021-10-03T09:20:32.856474 #1164166]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2021-10-03T09:20:32.856489 #1164166]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2021-10-03T09:20:32.856505 #1164166]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-10-03T09:20:32.856517 #1164166]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2021-10-03T09:20:32.856532 #1164166]  INFO -- :   Fusuma::Plugin::Inputs::AppmatcherInput
I, [2021-10-03T09:20:32.856546 #1164166]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2021-10-03T09:20:32.856560 #1164166]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2021-10-03T09:20:32.856574 #1164166]  INFO -- :   Fusuma::Plugin::Parsers::AppmatcherParser
I, [2021-10-03T09:20:32.856588 #1164166]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2021-10-03T09:20:32.856600 #1164166]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2021-10-03T09:20:32.856615 #1164166]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2021-10-03T09:20:32.856629 #1164166]  INFO -- : ---------------------------------------------
```

</details>

Could you paste the output of `fusuma --version`?

jJit0, 2021-10-03:
Here it is + my config file

```
W, [2021-10-03T17:22:20.264164 #4989]  WARN -- : config file: /root/.config/fusuma/config.yml is NOT FOUND
I, [2021-10-03T17:22:20.264209 #4989]  INFO -- : reload config: /usr/app/lib/ruby/gems/3.0.0/gems/fusuma-2.1.0/lib/fusuma/config.yml
W, [2021-10-03T17:22:20.375635 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.379836 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.384023 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.389589 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.394239 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.419038 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.423475 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.428224 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.432919 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.434154 #4989]  WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.439412 #4989]  WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.439686 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.445727 #4989]  WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.445846 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.450630 #4989]  WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.450749 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.455145 #4989]  WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.455236 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.461579 #4989]  WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.461673 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.465913 #4989]  WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.466008 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.470849 #4989]  WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.471035 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.476106 #4989]  WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.476258 #4989]  WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.483095 #4989]  WARN -- : fusuma-plugin-sendkey 0.6.2 is incompatible with running fusuma 2.1.0
W, [2021-10-03T17:22:20.487601 #4989]  WARN -- : fusuma-plugin-sendkey 0.6.2 is incompatible with running fusuma 2.1.0
I, [2021-10-03T17:22:20.487679 #4989]  INFO -- : ---------------------------------------------
I, [2021-10-03T17:22:20.487911 #4989]  INFO -- : Fusuma: 2.1.0
I, [2021-10-03T17:22:20.489878 #4989]  INFO -- : libinput: 1.19.1
I, [2021-10-03T17:22:20.489914 #4989]  INFO -- : ruby 3.0.0p0
I, [2021-10-03T17:22:20.491027 #4989]  INFO -- : OS: Linux 5.10.0-kali9-amd64 #1 SMP Debian 5.10.46-4kali1 (2021-08-09)
I, [2021-10-03T17:22:20.492059 #4989]  INFO -- : Distribution: Kali GNU/Linux Rolling \n \l
I, [2021-10-03T17:22:20.492960 #4989]  INFO -- : Desktop session: gnome x11
I, [2021-10-03T17:22:20.492990 #4989]  INFO -- : ---------------------------------------------
I, [2021-10-03T17:22:20.493005 #4989]  INFO -- : Enabled Plugins: 
I, [2021-10-03T17:22:20.493128 #4989]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-10-03T17:22:20.493141 #4989]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2021-10-03T17:22:20.493149 #4989]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-10-03T17:22:20.493156 #4989]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-10-03T17:22:20.493163 #4989]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-10-03T17:22:20.493175 #4989]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2021-10-03T17:22:20.493184 #4989]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-10-03T17:22:20.493192 #4989]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-10-03T17:22:20.493202 #4989]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-10-03T17:22:20.493214 #4989]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-10-03T17:22:20.493223 #4989]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-10-03T17:22:20.493231 #4989]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
[... cut ...]
    left:
      sendkey: 'LEFTALT+TAB'
    right:
      sendkey: 'LEFTALT+TAB'
    up:
      sendkey: 'LEFTMETA+S'
      keypress:
        LEFTSHIFT:
[... 23 lines cut ...]
  4:
    sendkey: 'LEFTALT+F11'

pinch:
  2:
    in:
      sendkey: 'LEFTCTRL+LEFTSHIFT+EQUAL'
    out:
      sendkey: 'LEFTCTRL+MINUS'

threshold:
  swipe: 0.1
  pinch: 0.1
  
interval:
  swipe: 0.5
  pinch: 0.5

plugin:
  inputs:
    libinput_command_input:
      show-keycodes: true
      enable-tap: true
      verbose: true
```

iberianpig, 2021-10-03:
Please see first 2 lines. 
It seems that fusuma loads default config from gem, not your conf.

> W, [2021-10-03T17:22:20.264164 #4989]  WARN -- : config file: /root/.config/fusuma/config.yml is NOT FOUND
I, [2021-10-03T17:22:20.264209 #4989]  INFO -- : reload config: /usr/app/lib/ruby/gems/3.0.0/gems/fusuma-2.1.0/lib/fusuma/config.yml

jJit0, 2021-10-03:
It's fixed Yamada-san

After upgrading ```gem update``` I got all fusuma plugins updated. It's working all working now

The ouput of  ```--version``` was the answer to my issues :yum: 

```
WARN -- : fusuma-plugin-tap 0.4.1 is incompatible with running fusuma 2.1.0
WARN -- : fusuma-plugin-keypress 0.4.1 is incompatible with running fusuma 2.1.0
WARN -- : fusuma-plugin-sendkey 0.6.2 is incompatible with running fusuma 2.1.0
```

closed 2021-10-03 by jJit0