iberianpig/fusuma #247: Can't detect gesture while using keyboard

DedSecer, 2021-10-03:
I find thay fusuma can't detect gesture while using keyboard, is there a option to enable gesture while using keyboard?

iberianpig, 2021-10-03:
Have you tried fusuma-plugin-keypress ?

Fusuma can detect gestures while pressing keys.
I think that keys you pressed and keys fusuma sent are mixed.
Maybe the OS doesn't recognize it as a shortcut key because some keys are being sent at the same time?

DedSecer, 2021-10-03:
Here is my config.yml
```
pinch:
  2:
    in:
      command: 'xdotool key ctrl+equal'
      # command: 'xdotool click 4'
      # sendkey: 'KP4'
    out:
      command: 'xdotool key ctrl+minus'
      # command: 'xdotool click 5'
      # sendkey: 'KP5'
  4:
    in:
      command: 'xdotool key super+a'
    out:
      command: 'xdotool key super+s'
```
My Operating System: Arch Linux
My Desktop: KDE

While I am typing and tyring to pinch in the web browser, I can't not zoom in or zoom out the page. I have installed the fusuma-plugin-keypress but it still doesn't work.

iberianpig, 2021-10-03:
> My Desktop: KDE

Which are you using Wayland or X11?

If using Wayland, could you try following config?

```yaml
pinch:
  2:
    in:
      sendkey: 'LEFTCTRL+EQUAL'
      threshold: 0.1
    out:
      sendkey: 'LEFTCTRL+MINUS'
      threshold: 0.1

```

DedSecer, 2021-10-04:
I'm using Xorg and my DE is KDE.

iberianpig, 2021-10-04:
> Sorry, I'm using Xorg.
Thanks👍

Please paste output of `fusuma --version` .

1. Have you tried that xdotool works without fusuma on your browser?
2.  Without while pressing any key,  gestures works correctly?

DedSecer, 2021-10-04:
My fusuma version is `	2.0.5-1` (install from archlinux AUR)

Xdotool cat zoom in or zoom out without fusuma in the web browser.

gesture work well without pressing keys.

closed 2022-04-28 by stale[bot]