iberianpig/fusuma #249: Allow more advanced gestures

MiguelGuthridge, 2021-10-09:
**Is your feature request related to a problem? Please describe.**
I'm currently considering switching to Linux and I need a replacement for the tool [GestureSign](https://gesturesign.win/#/), which implements similar functionality. One thing that it does that Fusuma does not, however, is allow for more complex gestures, such as multi-directional swipes (up then left, for example).

**Describe the solution you'd like**
It's be awesome to have support for these sorts of compound gestures, as well as diagonal ones. This would seriously increase the number of controls I could set up with it, which would be really awesome!

**Describe alternatives you've considered**
I can't think of any other than not adding the feature.

**Additional context**
Add any other context about the feature request here.

iberianpig, 2021-10-09:
It is interesting.

It may be a little difficult with the current code base.
It is necessary to wait the multiple gestures detection as much as possible so as not to trigger a single gesture.

MiguelGuthridge, 2021-10-10:
If I had the time, I'd love to try, but I doubt I will

closed 2022-04-28 by stale[bot]