iberianpig/fusuma #258: Autostart Fusuma on KDE

LeonardoMantovani, 2021-11-06:
**Is your feature request related to a problem? Please describe.**
The README is very exhaustive regarding the installation and the configuration of Fusuma, but is missing how to set up autostart for non-GNOME desktop environments, which in my opinion is very useful

**Describe the solution you'd like**
Instructions on how to autostart Fusuma on the main README file (below the GNOME instructions that already exists)  

**Describe alternatives you've considered**
I tried creating a .sh file with the following text and adding it to autostart settings, but nothing happens after system reboot (and I need to run my `fusuma.sh` manually)
```
#!/usr/bin/env bash
fusuma -d
```

iberianpig, 2021-11-07:
Basically, I don't think autostart configuration is difficult.

Isn't it almost the same as what we do here?
https://github.com/iberianpig/fusuma#autostart-gnome-session-properties

Did you check the path with `$ which fusuma`?
Try to set that path in your .sh file.

LeonardoMantovani, 2021-11-08:
Yeah I found out that the problem with my procedure was the missing path to fusuma.
using the output of `which fusuma` in the `.sh` file worked.

closed 2021-11-08 by LeonardoMantovani