iberianpig/fusuma #259: Bring back Tap and hold Gestures !

HritwikSinghal, 2021-11-10:
**Is your feature request related to a problem? Please describe.**
Now that Libinput (with release of 1.18.0) has official API for Hold gestures (and i think for tap as well), see [This](https://lists.x.org/archives/wayland-devel/2021-September/041977.html), can we get the tap and hold gestures back?

iberianpig, 2021-11-10:
Yes, I have checked libinput Hold gestures API and its logs.
I'm planning to import only the Hold gestures from fusuma-plugin-tap into Fusuma as a default plugin. However, I don't plan to import tap gestures because they are still unstable.

HritwikSinghal, 2021-11-10:
> However, I don't plan to import tap gestures because they are still unstable.

So are They going to be stable in future or is there no work in bringing them from libinput devs? i cant find anything on their gitlab so thats why  i am asking

iberianpig, 2021-11-13:
libinput debug-events does not report tap gestures. Currently, it is only possible to observe the tap state via the unstable --verbose log.

ref: https://gitlab.freedesktop.org/libinput/libinput/-/blob/1.19.0/src/libinput.h#L971-981

Swivelgames, 2022-06-29:
Is this really `wontfix`? I believe it's still in the README. Might just be `stalebot` mistakenly marking things? (hopefully :grimacing: )