iberianpig/fusuma #262: [Gesture]"Pinch in & Pinch out" does not work most of the time <Arch><KDE><Surface>

2ndBillGates, 2021-12-02:
**Describe the bug**
Cannot use 3/4 finger pinch in/ out. 3/4 finger Rotate sometimes work.
Pinch in & out works out well on other programs like Libinput-gesture // touchegg

I am using Microsoft Surface Book 2, Arch Linux, KDE Plasma

BTW fusuma-sendkey does not work for me too. When I run  fusuma-sendkey LEFTALT+RIGHT 
nothing happened. When I insert command into fusuma config, it does not work as well, it show LEFTALT+RIGHT output, but no key is pressed.

**To Reproduce**
Steps to reproduce the behavior:
1. Pinch in or Pinch Out
2. Fusuma debug shows no output
3. libinput debug-events does detect Pinch 

Left: libinput debug-events Right: fusuma 
[image]
In the picture shows that Pinch gesture has detect multiple times but fusuma does nothing.

**Expected behavior**
It works on other Gesture program like, ' libinput-gesture, touchegg....' 

I also would like to see future support for  GUI & "MacOS three finger drag" without using Xdotool press down. Because it keep request and stress CPU.
Arch packages "libinput-three-finger-drag" seems to does a pretty good job on drag.

**Versions**
```yml

Fusuma: 2.2.0
libinput: 1.19.2
ruby 3.0.2p107
```

**Configuration**
```yml
# ========= My version ==========

swipe:
  #3:
    #begin:
      #command: xdotool mousedown 1
    #update:
      #command: xdotool mousemove_relative -- $move_x, $move_y
      #interval: 0.01
      #accel: 2
    #end:
      #command: xdotool mouseup 1

  4:
    left:
#      workspace: 'next' # Switch to next workspace
#      command: 'xdotool key --clearmodifiers super+ctrl+Right'  # Move next Desktop
      command: 'xdotool key alt+Right' # History forward
#      sendkey: "LEFTALT+RIGHT" # history back
      keypress:
        LEFTSHIFT:
          window: 'next' # Move window to next workspace
        LEFTMETA:
          command: 'xdotool key --clearmodifiers super+ctrl+Left' # Move window to left side

    right:
#      workspace: 'prev' # Switch to previous workspace
#      command: 'xdotool key --clearmodifiers super+ctrl+Left'  # Move to Left Desktop
      command: 'xdotool key alt+Left' # History back
#      sendkey: "LEFTALT+LEFT" # history forward
      keypress:
        LEFTSHIFT:
          window: 'prev' # Move window to previous workspace
        LEFTMETA:
          command: 'xdotool key --clearmodifiers super+ctrl+Right'  # Move window to right side
    up:
#      command: 'xdotool key Control_L+F10' # Workspace overview
      workspace: 'next' # Switch to next workspace
#      command: 'xdotool key --clearmodifiers alt+Right'  # go to Previous
#      command: 'xdotool key --clearmodifiers super+ctrl+Down'  # Move to Down Desktop
      keypress:
        LEFTMETA:
          window:
            maximized: 'toggle' # Toggle Maximize/Unmaximize Window
    down:
      command: 'xdotool key Shift_L+Alt_L+Tab' #aaaaa
      workspace: 'prev' # Switch to next workspace
      #command: 'xdotool key --clearmodifiers alt+Right'  # go to Next
      #command: 'xdotool key --clearmodifiers super+ctrl+Up'  # Move to Up Desktop
      keypress:
        LEFTMETA:
          window: 'close' # Close window

#pinch:
  ##2:
    ##in:
      ##command: 'xdotool key Control_L+F10' # Zoom in
    ##out:
      ##command: 'xdotool key super+d' # Zoom out
  ##3:
    ##in:
      ##command: 'xdotool keydown ctrl click 4 keyup ctrl' # Zoom in
    ##out:
      ##command: 'xdotool keydown ctrl click 5 keyup ctrl' # Zoom out
  #3:
    #in:
      #command: 'xdotool key Super_L+Tab' # Window overview
    #out:
      #command: 'xdotool key --clearmodifiers alt+shift+F1' # Workspace overview

  #4:
    #in:
      #command: 'xdotool key Super_L+Tab' # Window overview
    #out:
      #command: 'xdotool key --clearmodifiers alt+shift+F1' # Workspace overview

pinch:
  2:
    in:
      command: "echo 2\ finger\ pinch\ in"
    out:
      command: "echo 2\ finger\ pinch\ out"
  3:
    in:
      command: "echo 3\ finger\ pinch\ out"
    out:
      command: "echo 3\ finger\ pinch\ out"
  4:
    in:
      command: "echo 4\ finger\ pinch\ in"
    out:
      command: "echo 4\ finger\ pinch\ out"

#rotate:
  #3:
    #clockwise:
      #command: 'xdotool key XF86MonBrightnessUp' # Brightness up
    #counterclockwise:
      #command: 'xdotool key XF86MonBrightnessDown' # Brightness down

#rotate:
  #3:
    #clockwise:
      #command: 'xdotool key Super_L+Tab' # Window overview
    #counterclockwise:
      #command: 'xdotool key --clearmodifiers alt+shift+F1' # Workspace overview
  #4:
    #clockwise:
      #command: 'xdotool key Super_L+Tab' # Window overview
    #counterclockwise:
      #command: 'xdotool key --clearmodifiers alt+shift+F1' # Workspace overview

plugin:
  inputs:
    libinput_command_input: # options for lib/plugin/inputs/libinput_command_input
      enable-tap: true # click to tap
      enable-dwt: true # disable tap while typing
      show-keycodes: true # https://github.com/iberianpig/fusuma-plugin-keypress#add-show-keycode-option
      #executors:
        #sendkey_executor:
          #device_name: 'Microsoft Surface Keyboard'

```

iberianpig, 2023-03-25:
Note. The reason pinch operation is not recognized well may be the same as following Issue.
https://github.com/iberianpig/fusuma/issues/291

iberianpig, 2025-04-02:
Please adjust the pinch threshold. If the issue still persists after making that change, please feel free to reopen the issue.

closed 2025-04-02 by iberianpig