iberianpig/fusuma #263: Gestures on Wayland

musaoruc, 2021-12-06:
Gestures do not work in applicaions running natively on wayland in gnome. Fusuma registers the gestures (can be seen in the output) but they do not work. They work on X11 session or applications running on xwayland.

Steps to reproduce the behavior:
1. Open an application running on wayland
2. Try to use the gesture
3. Gesture does not work

**Expected behavior**
Gestures should work on wayland

**Versions**
```yml
I, [2021-12-06T13:08:15.562018 #20796]  INFO -- : reload config: /home/musaoruc/.config/fusuma/config.yml
I, [2021-12-06T13:08:15.740229 #20796]  INFO -- : ---------------------------------------------
I, [2021-12-06T13:08:15.740902 #20796]  INFO -- : Fusuma: 2.1.0
I, [2021-12-06T13:08:15.745811 #20796]  INFO -- : libinput: 1.19.2
I, [2021-12-06T13:08:15.745942 #20796]  INFO -- : ruby 3.0.2p107
I, [2021-12-06T13:08:15.747736 #20796]  INFO -- : OS: Linux 5.15.4-zen1-1-zen #1 ZEN SMP PREEMPT Sun, 21 Nov 2021 21:34:35 +0000
I, [2021-12-06T13:08:15.749200 #20796]  INFO -- : Distribution: Arch Linux \r (\l)
I, [2021-12-06T13:08:15.754949 #20796]  INFO -- : Desktop session: gnome wayland
I, [2021-12-06T13:08:15.755061 #20796]  INFO -- : ---------------------------------------------
I, [2021-12-06T13:08:15.755088 #20796]  INFO -- : Enabled Plugins: 
I, [2021-12-06T13:08:15.755320 #20796]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-12-06T13:08:15.755345 #20796]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2021-12-06T13:08:15.755363 #20796]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2021-12-06T13:08:15.755385 #20796]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2021-12-06T13:08:15.755405 #20796]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2021-12-06T13:08:15.755425 #20796]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-12-06T13:08:15.755445 #20796]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-12-06T13:08:15.755463 #20796]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-12-06T13:08:15.755481 #20796]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2021-12-06T13:08:15.755499 #20796]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2021-12-06T13:08:15.755517 #20796]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-12-06T13:08:15.755537 #20796]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-12-06T13:08:15.755555 #20796]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2021-12-06T13:08:15.755575 #20796]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-12-06T13:08:15.755593 #20796]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-12-06T13:08:15.755612 #20796]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2021-12-06T13:08:15.755631 #20796]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-12-06T13:08:15.755649 #20796]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2021-12-06T13:08:15.755669 #20796]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2021-12-06T13:08:15.755687 #20796]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2021-12-06T13:08:15.755707 #20796]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2021-12-06T13:08:15.755726 #20796]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2021-12-06T13:08:15.755745 #20796]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2021-12-06T13:08:15.756030 #20796]  INFO -- : ---------------------------------------------

```

**Configuration**
```yml
swipe:
  3:
    left:
      command: "xdotool key alt+Right" # History forward
    right:
      command: "xdotool key alt+Left" # History back

```

iberianpig, 2021-12-06:
Please try https://github.com/iberianpig/fusuma-plugin-sendkey

musaoruc, 2021-12-06:
Thank you so much! That worked.

closed 2021-12-06 by musaoruc