iberianpig/fusuma #264: Overriding pop os touchpad gestures with fusuma

tae898, 2021-12-23:
Hi,

I installed pop os 21.10, [which comes with its multi-touch touchpad gestures](https://www.techrepublic.com/article/pop-os-21-04-has-landed-and-its-out-of-this-world/), unlike native Ubuntu.

I don't know what exactly is running in the backend, and also I don't know how to modify them. So I just wanted to use fusuma, as I did with native Ubuntu. But when I run fusuma, it was running both pop os gestures and fusuma gestures at the same time. I tried this and that, and what worked in the end was to:

1. [Install X11 Gestures](https://extensions.gnome.org/extension/4033/x11-gestures/)
2. Uninstall it.
3. Run fusuma

Above made it possible for me to only use fusuma. This is a hack, and not a proper way. I wonder if you have a better way to override pop os multi-touch gestures with fusuma.

Thanks!

iberianpig, 2021-12-24:
Thank you for reporting.

I think it is hard to control Pop!_OS 21.10's touchegg or Gnome Wayland's built-in gesture settings from Fusuma.
I don't know if it can be possible, the /autostart/touchegg.desktop file or systemd service should be set to auto-start touchegg.
You can turn off that auto-start.

tae898, 2021-12-24:
hi thanks for getting back to me.

Do you also have an idea how and why my "hack" worked?

iberianpig, 2021-12-25:
No, I don't. 
I wonder if the configuration overwritten at the time of installation are deleted or disabled by uninstallation.

closed 2025-07-08 by iberianpig