iberianpig/fusuma #266: I have created a little tutorial on how to Start fusuma on Login in KDE

Shadowphoenix, 2022-01-26:
Since there is a tutorial for Gnome, might as well add one for KDE ;P

1. run `whereis fusuma` - this should default to `~/.local/share/gem/ruby/3.0.0/bin/fusuma` or `/home/$USER/.local/share/gem/ruby/3.0.0/bin/fusuma`
2. create a script that looks like this:
[image]
3. make it executable in Fileproperties (by Ticking the corresponding box):
[image]
4. Go to KDE-Settings, search for "Autostart".
5. Click on "Add" and Add your Script as a Loginscript.
6. Reboot
Done :)

Terminology might slightly differ depending on selected Locale

Shadowphoenix, 2022-05-02:
(activity to keep active)

iberianpig, 2022-05-04:
Hi, thanks for your suggestion.
Does it use ~/.config/autostart/ in KDE?

Shadowphoenix, 2022-05-06:
> Hi, thanks for your suggestion. Does it use ~/.config/autostart/ in KDE?

yes it does.

DocteurFlashbios, 2023-01-12:
@iberianpig  Thank you very much for this work !! I can't do without ;)
The autostart is a great idea, but I don't understand the point of creating a bash script.
I tried adding directly the path in KDE settings, and the "-d" in the "arguments" input...
... seems working smooth ! Not for you ?

iberianpig, 2023-01-13:
I don't think there is a need to create a bash script unless there is a particular reason such as wanting to do something before or after. It is important to get the path where the actual fusuma is located using the `which` command.

Shadowphoenix, 2023-03-01:
> @iberianpig Thank you very much for this work !! I can't do without ;) The autostart is a great idea, but I don't understand the point of creating a bash script. I tried adding directly the path in KDE settings, and the "-d" in the "arguments" input... ... seems working smooth ! Not for you ?

turns out within kde you can just run the command `fusuma` directly as an application to be executed on launch of plasmashell. i didnt know that at the time and forgot to update when i found out

Shadowphoenix, 2023-03-01:
this works perfectly fine for me, once $PATH for $SHELL is configured to contain gems

Shadowphoenix, 2023-03-12:
Turns out the builtin Gestures on KDE afaik only work on Wayland but they are set to Mac-Esque by default so i will likely ditch fusuma because it is no longer needed for me. Thanks for your work on this tho, @iberianpig .

closed 2023-03-12 by iberianpig