iberianpig/fusuma #267: Update to 2.3.0, strange behaviour on Wayland

SamuXzX, 2022-01-30:
**Describe the bug**
I first had some problems with 4 fingers gestures, I found that I didn't update fusuma for a long time and I update it. 4 fingers gestures solved, now everything else doesn't work.
I start `fusuma` from command line, now it doesn't write what gesture I'm executing even if the gesture properly fires: with a 4 fingers horizontal swipe I can change desktops, but the vertical gestures don't work (they don't execute the right shortcut).
In addition, any of the 3 finger gestures works.
I see from `fusuma --version` that fusuma-sendkey isn't compatible with this version, and that fusuma-2.3.0 doesn't even show in the changelogs. Could it be because of this? What am I doing wrong

**To Reproduce**
Steps to reproduce the behavior:
1. Start `fusuma` in command line
2. Test all gestures

**Expected behavior**
All gestures should fire the proper shortcut as in X11.

**Versions**
```yml
I, [2022-01-30T10:54:16.047166 #6040]  INFO -- : reload config: /home/samuele/.config/fusuma/config.yml
fatal: .git non è un repository Git (né lo è alcuna delle directory genitrici)
W, [2022-01-30T10:54:16.178367 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.191083 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.203900 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.209562 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.216047 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.241020 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.246846 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.253203 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.260401 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.267628 #6040]  WARN -- : fusuma-plugin-tap 0.3.2 is incompatible with running fusuma 2.3.0
fatal: .git non è un repository Git (né lo è alcuna delle directory genitrici)
W, [2022-01-30T10:54:16.283024 #6040]  WARN -- : fusuma-plugin-sendkey 0.5.1 is incompatible with running fusuma 2.3.0
W, [2022-01-30T10:54:16.289335 #6040]  WARN -- : fusuma-plugin-sendkey 0.5.1 is incompatible with running fusuma 2.3.0
I, [2022-01-30T10:54:16.289673 #6040]  INFO -- : ---------------------------------------------
I, [2022-01-30T10:54:16.290707 #6040]  INFO -- : Fusuma: 2.3.0
I, [2022-01-30T10:54:16.297134 #6040]  INFO -- : libinput: 1.19.1
I, [2022-01-30T10:54:16.297313 #6040]  INFO -- : ruby 2.7.0p0
I, [2022-01-30T10:54:16.300174 #6040]  INFO -- : OS: Linux 5.13.0-27-generic #29~20.04.1-Ubuntu SMP Fri Jan 14 00:32:30 UTC 2022
I, [2022-01-30T10:54:16.303264 #6040]  INFO -- : Distribution: KDE neon 5.23 \n \l
I, [2022-01-30T10:54:16.306728 #6040]  INFO -- : Desktop session: plasmawayland wayland
I, [2022-01-30T10:54:16.306959 #6040]  INFO -- : ---------------------------------------------
I, [2022-01-30T10:54:16.307036 #6040]  INFO -- : Enabled Plugins: 
I, [2022-01-30T10:54:16.307431 #6040]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2022-01-30T10:54:16.307511 #6040]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2022-01-30T10:54:16.307578 #6040]  INFO -- :   Fusuma::Plugin::Detectors::HoldDetector
I, [2022-01-30T10:54:16.307642 #6040]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2022-01-30T10:54:16.307710 #6040]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2022-01-30T10:54:16.307775 #6040]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2022-01-30T10:54:16.307840 #6040]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2022-01-30T10:54:16.307904 #6040]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2022-01-30T10:54:16.307971 #6040]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2022-01-30T10:54:16.308036 #6040]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2022-01-30T10:54:16.308102 #6040]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2022-01-30T10:54:16.308167 #6040]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2022-01-30T10:54:16.308232 #6040]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2022-01-30T10:54:16.308297 #6040]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2022-01-30T10:54:16.308365 #6040]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2022-01-30T10:54:16.308478 #6040]  INFO -- : ---------------------------------------------
```

**Configuration**
```yml
swipe:
 3:
  left:
   sendkey: "LEFTMETA+LEFT" # Compact to the left

  right:
   sendkey: "LEFTMETA+RIGHT" # Compact to the right

  down:
   sendkey: "LEFTMETA+A" # Present windows

  up:
   sendkey: "LEFTMETA+UP" # Toggle maximization

 4:
  left:
   sendkey: "LEFTMETA+TAB" # Go one desktop right

  right:
   sendkey: "LEFTMETA+LEFTSHIFT+TAB" # Go one desktop left

  down:
   sendkey: "LEFTMETA+LEFTSHIFT+A" # Show desktops grid

  up:
   sendkey: "LEFTALT+F4" # Close window
  
pinch:
 in:
  sendkey: "LEFTCTRL+KPPLUS" # Zoom in

 out:
  sendkey: "LEFTCTRL+KPMINUS" # Zoom out

threshold:
 pinch: 0.75
```

iberianpig, 2022-01-30:
Hi! Could you try to update fusuma-plugins with following commands?

```sh
sudo gem list fusuma-plugin- | cut -d' ' -f1 | xargs --no-run-if-empty sudo gem update
```

SamuXzX, 2022-01-30:
Thanks, something changed, but not much.
At first the 4 fingers gesture to the right worked a couple of times, but then stopped working and any of the 3 or 4 fingers gestures works. The pinch gestures works consistently.

[image]

SamuXzX, 2022-02-03:
I think I found the problem: the trackpad sensitiveness was drastically reduced once I update fusuma. Maybe the default value has changed along with the versions? I simply lowered the threshold and now it works.

korason7117, 2022-04-17:
> Hi! Could you try to update fusuma-plugins with following commands?
> 
> ```shell
> sudo gem list fusuma-plugin- | cut -d' ' -f1 | xargs --no-run-if-empty sudo gem update
> ```

Now that i've tried you command, Fusuma behaves weirdly.
No matter if I delete the gestures for 4 Figer-Gestures or whatever edit I do to 4 Finger-Gestures is always the same 4 Finger-Gesures.
Also, i dont seem to find any related process in htop by typing `fusuma`.
Killing Fusuma still keeps alive your suggested command, "your suggested command" keeps alive only the 4 finger-gestures. While the 3 finger-gestures die along with the Fusuma process

NOTE: your suggested command persist even after rebooting, how do I kill that process?

iberianpig, 2022-04-17:
@SamuXzX I don't have Plasma Wayland environment.
I want to know if fusuma-sendkey works completely on Plasma Wayland.
Could you confirm with `fusuma-sendkey` command works well on your terminal?

@kosaron7117

I don't know your environment or settings.
Could you paste your output of `$ fusuma --version` ?

 > how do I kill that process?

Could you `grep` the output of `ps aux`?
If you get fusuma process ID, then send a kill signal.
```
iberianpig@iberianpig-XPS-13-9310:~$ ps aux | grep fusuma
iberian+    4178  0.3  0.1 509388 27396 ?        Sl    4月12  28:14 /home/iberianpig/.rbenv/versions/3.0.2/bin/ruby /home/iberianpig/.rbenv/versions/3.0.2/bin/fusuma -d --log=/tmp/fusuma.log --verbose
iberian+    4181  0.0  0.0 100088 14760 ?        S     4月12   6:55 fusuma/plugin/inputs/timer_input
iberian+    4185  0.0  0.0 232504 11008 ?        Sl    4月12   0:00 fusuma/plugin/appmatcher/user_switcher
iberian+  332419  0.0  0.0  10096  2580 pts/15   R+   23:01   0:00 grep --color=auto fusuma
iberianpig@iberianpig-XPS-13-9310:~$ kill 4178
```

korason7117, 2022-04-17:
@iberianpig The output of fusuma from running on terminal and config contents of `config-kde.yml`: https://privatebin.net/?2b349326ffea69f0#9C7SsFu2qi1geMwemi7CRfEaQwoJVb9KJNpwd5d2qm5g

When running `ps aux | grep fusuma` with `fusuma -d`
```
sh-5.1$ ps aux | grep fusuma
okama       2941  1.0  0.1 285432 16220 ?        Sl   12:12   0:11 /usr/bin/ruby /home/okama/.local/share/gem/ruby/3.0.0/bin/fusuma -c /home/okama/.config/fusuma/config-kde.yml -d
okama       2945  0.0  0.1 150120 15636 ?        S    12:12   0:00 fusuma/plugin/inputs/timer_input
okama       3432  0.0  0.0   6672  2660 pts/1    S+   12:29   0:00 grep fusuma
```
Running `ps aux | grep fusuma` with the above processes killed :
```
sh-5.1$ ps aux | grep fusuma
okama       3645  0.0  0.0   6672  2636 pts/1    S+   12:41   0:00 grep fusuma
```
Funny, I dont know what your command did to the core files for fusuma to behave like this.
By the way, I ran `sudo gem list fusuma-plugin- | cut -d' ' -f1 | xargs --no-run-if-empty sudo gem update`
without sudo. Also your command did work and fixed the weird issue of 4 Finger-Gestures,
it felt like the `interval` and `threshold` were at 0 meaning, way to fast. making it impossible to move around.
The reason why that speed was out of control, was because I still had the 4 Finger-Gestures in the config file.
I had to disable them for the 4 Finger-Gestures to have a normalized speed.
But. in the end, now I have no way to modify the 4 Finger-Gestures, because I can't find the process and kill it

SamuXzX, 2022-04-19:
> @SamuXzX I don't have Plasma Wayland environment. I want to know if fusuma-sendkey works completely on Plasma Wayland. Could you confirm with `fusuma-sendkey` command works well on your terminal?

It does. My LEFTMETA key opens the overview:

https://user-images.githubusercontent.com/30182251/164015563-3a8e5404-527a-45b3-9598-ac20d7a54561.mp4

What doesn't seem to work are the actual gestures. For example, for me LEFTMETA+TAB goes a desktop to the right. It goes well if I use it from the command line. But, as you can see [here](https://nextcloud.samuelezappala.it/s/yBKnyRTDGxSX4oE), when I try to use its gestures it kind of flickers back and forth.

korason7117, 2022-04-19:
@SamuXzX try disabling 4 Finger-Gestures.
im seeing that you have 4 Finger-Gestures activated, meaning that the `interval` and `threshold` are colliding
with the default 4 Finger-Gestures on KDE-Plasma-Wayland
It worked for me. I had to disable them, and I have to bear with these default gestures because KDE-Plasma devs hard coded 4 Finger Gestures into KDE-Plasma on Wayland
> 
>     * By default on Wayland: 4 Finger-Gestures;
> 
>     * Swipe-Down: Toggle Preset Windows (All desktops) Ctrl+F10 (default keybinds)
> 
>     * Swipe-Up: Desktop Grid Ctrl+F8 (default keybind)
> 
>     * Swipe-Right/Left: Switch to Next/Previous Desktop
NOTE: I could be wrong, and be spreading misinformation about KDE-Plasma hard coding all this on Wayland, but
I couldn't find any relevant info in KDE-Plasma Changelogs about TouchPad-Gestres.

SamuXzX, 2022-04-19:
I disabled 4 fingers gestures on Wayland and I was able to use those hard coded in KDE Plasma. Now it seems obvious they were obstructing each other, rather than "conflicting", as both were working at the same time.

So, we simply discovered that there were already some not-yet configurable gestures on KDE Plasma Wayland. I'd say to find a way to bypass them, because otherwise Fusuma won't have an usecase. But they are not configurable, so it will take some time until this could happen.

korason7117, 2022-04-19:
> So, we simply discovered that there were already some not-yet configurable gestures on KDE Plasma Wayland. I'd say to find a way to bypass them

Is it even possible to bypass a hard coded setting from KDE-Plasma? Best way as of now would be
to compile KDE-Plasma from source and do your own tweaks.

I've tried to add **Swipe** `modifiers` like this:

```
4:
 keypress:
  LEFTCTRL:
   sendkey: 'LEFTCTRL+LEFTSHIFT+LEFT' #Tile window to the left
```
And no way around it, It works as a gesture, but Wayland default Gesture triggers too.
Making modifiers for **Swipe** 4 Finger-Gestures unusable.
Gestures such as `pinch`, `rotate` with `modifiers` for 4 Finger-Gestures works as intended

SamuXzX, 2022-04-20:
I mean: now it surely isn't possible. Maybe in the future, when KDE-Plasma gesture will receive a configuration page, it will be. But it doesn't matter: in such a case it would be possible to disable them to give space to Fusuma.

I think that at this point the issue can be closed and we could add to the README that, in KDE-Plasma Wayland, Fusuma will collide with the hard coded gestures.

iberianpig, 2022-10-07:
The situation is the same in Gnome Wayland. Wayland reads libinput on the compositor side, but Fusuma also reads from libinput-debug-inputs and treats them as gestures. Therefore, Gnome Wayland and Fusuma execute actions at the same time. There are currently two workarounds
1. Configure Fusuma so that Wayland native gestures do not conflict
2. Disable Wayland native gestures and configure gestures in Fusuma only

SamuXzX, 2022-10-07:
How is (1) even possible? Do you simply mean "stopping using the same gestures as Wayland does"?
The workaround (2) seems totally in power of the distroes. Maybe, other than just waiting for the distroes to let to configure their gestures, we can only contact them to request some sort of API to let Fusuma and other services take charge of the gestures for the system.
It seems to me that now that distroes are equipping themselves with gestures, Fusuma can only be an alternative, so that we have to wait for them to let users disable native gestures.

pier-bezuhoff, 2025-11-27:
For GNOME 47,48,49 on Wayland there is [this extension](https://extensions.gnome.org/extension/8830/disable-3-finger-gestures-redux/) that suppresses native GNOME 3-finger gestures, which allows fusuma to work as expected.
Additionally, to disable native 4-finger gestures you may want to adjust its `extension.js` L37
`return real_count == 3 ? 0 : real_count;`
to
`return real_count >= 3 ? 0 : real_count;`
This workaround did it for me at least.

iberianpig, 2026-10-03:
The discussion indicates that the reported behavior is caused by compositor-native gestures on Wayland running alongside Fusuma. Fusuma cannot disable or override hard-coded gestures in GNOME or KDE. The current workaround is to disable or avoid the conflicting compositor gestures, or configure different gesture combinations. This is not actionable in Fusuma itself, so I am closing this issue as not planned. Please open a new issue if a current Fusuma-specific regression can be reproduced after disabling the compositor gesture.

closed 2026-10-03 by iberianpig