iberianpig/fusuma #269: Fusuma doesn't recognize multi fingers swipe gestures

minhduc66532, 2022-03-20:
**Describe the bug**
Fusuma doesn't recognize multi fingers swipe gestures, haven't tried Wayland yet

**Versions**
```yml
I, [2022-03-20T17:26:49.895523 #44083]  INFO -- : reload config: /home/minhduc/.config/fusuma/config.yml
I, [2022-03-20T17:26:49.938597 #44083]  INFO -- : ---------------------------------------------
I, [2022-03-20T17:26:49.938836 #44083]  INFO -- : Fusuma: 2.3.0
I, [2022-03-20T17:26:49.939446 #44083]  INFO -- : libinput: 1.20.0
I, [2022-03-20T17:26:49.939475 #44083]  INFO -- : ruby 3.0.3p157
I, [2022-03-20T17:26:49.940422 #44083]  INFO -- : OS: Linux 5.16.12-xanmod1-1 #1 SMP Tue, 08 Mar 2022 15:07:13 +0000
I, [2022-03-20T17:26:49.940928 #44083]  INFO -- : Distribution: Arch Linux \r (\l)
I, [2022-03-20T17:26:49.943338 #44083]  INFO -- : Desktop session: kde x11
I, [2022-03-20T17:26:49.943441 #44083]  INFO -- : ---------------------------------------------
I, [2022-03-20T17:26:49.943466 #44083]  INFO -- : Enabled Plugins:
I, [2022-03-20T17:26:49.943632 #44083]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2022-03-20T17:26:49.943652 #44083]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2022-03-20T17:26:49.943666 #44083]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2022-03-20T17:26:49.943678 #44083]  INFO -- :   Fusuma::Plugin::Detectors::HoldDetector
I, [2022-03-20T17:26:49.943691 #44083]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2022-03-20T17:26:49.943705 #44083]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2022-03-20T17:26:49.943718 #44083]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2022-03-20T17:26:49.943731 #44083]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2022-03-20T17:26:49.943744 #44083]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2022-03-20T17:26:49.943756 #44083]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2022-03-20T17:26:49.943769 #44083]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2022-03-20T17:26:49.943782 #44083]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2022-03-20T17:26:49.943794 #44083]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2022-03-20T17:26:49.943808 #44083]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2022-03-20T17:26:49.943820 #44083]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2022-03-20T17:26:49.943834 #44083]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2022-03-20T17:26:49.943847 #44083]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2022-03-20T17:26:49.943859 #44083]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2022-03-20T17:26:49.943871 #44083]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2022-03-20T17:26:49.943885 #44083]  INFO -- : ---------------------------------------------
```

**Configuration**
```yml
swipe:
  2:
    left:
      sendkey: "LEFTALT+RIGHT" # history back
    right:
      sendkey: "LEFTALT+LEFT" # history forward
  
  3:
    left:
      sendkey: "PREVIOUSSONG"
    right:
      sendkey: "NEXTSONG"
    up:
      sendkey: "VOLUMEUP"
    down:
      sendkey: "VOLUMEDOWN"

tap:
  3:
    sendkey: "PLAYPAUSE"

plugin: 
  inputs:
    libinput_command_input: # options for lib/plugin/inputs/libinput_command_input
      enable-tap: true # click to tap
      enable-dwt: true # disable tap while typing
      show-keycodes: true # https://github.com/iberianpig/fusuma-plugin-keypress#add-show-keycode-option
      verbose: true

```
**Run fusuma -v from the terminal and try 2 fingers swiping gesture**
[... 128 lines cut ...]
-event6   DEVICE_ADDED            Logitech Gaming Keyboard G610 Keyboard seat0 default group7  cap:k
-event7   DEVICE_ADDED            Logitech G703 LS                  seat0 default group8  cap:kp left scroll-nat scroll-button
-event17  DEVICE_ADDED            USB2.0 HD UVC WebCam: USB2.0 HD   seat0 default group9  cap:k
-event18  DEVICE_ADDED            ELAN1203:00 04F3:307A Mouse       seat0 default group10 cap:p left scroll-nat scroll-button
-event19  DEVICE_ADDED            ELAN1203:00 04F3:307A Touchpad    seat0 default group10 cap:pg  size 101x52mm tap(dl off) left scroll-nat scroll-2fg-edge dwt-on
-event11  DEVICE_ADDED            Asus WMI hotkeys                  seat0 default group11 cap:k
-event4   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group12 cap:k
-event19  GESTURE_HOLD_BEGIN      +0.040s	2
 event19  GESTURE_HOLD_END        +0.048s	2 cancelled
 event19  POINTER_SCROLL_FINGER   +0.048s	vert 0.00/0.0 horiz -7.73/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.055s	vert 0.00/0.0 horiz -9.21/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.062s	vert 0.00/0.0 horiz -9.05/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.069s	vert 0.00/0.0 horiz -9.21/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.076s	vert 0.00/0.0 horiz -9.70/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.083s	vert 0.00/0.0 horiz -5.59/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.090s	vert 0.00/0.0 horiz -12.17/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.097s	vert 0.00/0.0 horiz -12.67/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.104s	vert 0.00/0.0 horiz -12.67/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.111s	vert 0.00/0.0 horiz -5.10/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.118s	vert 0.00/0.0 horiz -12.67/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.125s	vert 0.00/0.0 horiz -10.86/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.132s	vert 0.00/0.0 horiz -10.20/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.139s	vert 0.00/0.0 horiz -17.11/0.0* (finger)
 event19  POINTER_SCROLL_FINGER   +0.146s	vert 0.00/0.0* horiz 0.00/0.0* (finger)
```

iberianpig, 2022-03-20:
Fusuma doesn't support 2 fingers swipe.
This is a restrictions of libinput.

minhduc66532, 2022-03-22:
Oof can confirm I change the history gestures to 3 fingers and it working properly right now. Also is there any way to increase/decrease volume without swiping multiple time ? For example, on Window you can swipe continuously to change the volume accordingly

iberianpig, 2022-04-17:
The following config is a part of my `~/.config/fusuma/config.yml`.

The three-finger rotate gesture changes the volume; pressing the Shift key decreases the width of the volume change for fine-tuning. Hold down the ALT key while rotating to change the brightness.

see: https://twitter.com/nukumaro22/status/1393232055383691274

```yaml
# Volume and brightness adjustment by swiping/rotating

rotate:
  3:
    clockwise: # clockwise
      update:
        sendkey: VOLUMEUP # volume up 
        LEFTSHIFT: # hold down Shift
          sendkey: LEFTSHIFT+VOLUMEUP # volume up (slight increase)
        LEFTALT: # Hold down Alt
          sendkey: BRIGHTNESSUP # brightness up
    counterclockwise: # counterclockwise
      update:
        sendkey: VOLUMEDOWN # volume DOWN
        LEFTSHIFT: # Hold down Shift
          sendkey: LEFTSHIFT+VOLUMEDOWN # volume DOWN (slight decrease)
        LEFTALT: # Hold down Alt
          sendkey: BRIGHTNESSDOWN # brightness DOWN
```

minhduc66532, 2022-05-01:
Brilliant, thanks for sharing

closed 2022-05-01 by minhduc66532