iberianpig/fusuma #275: Modifier Keys

Swivelgames, 2022-06-29:
**Describe the solution you'd like**
It would be a game-changer to be able to combine gestures with modifier keys.

For instance, a function may be different depending on whether or not I'm holding down <kbd>Ctrl</kbd>, <kbd>Alt</kbd>, <kbd>Super</kbd>, <kbd>Shift</kbd>, or a combination of either of them.

I use i3 quite a bit, and my config is setup in a way where combinations of modifiers increase the significance of the action.

Navigating:
- <kbd>Super</kbd> and <kbd>Left</kbd>/<kbd>Right</kbd> -- Focus Window Left/Right
- <kbd>Super</kbd>+<kbd><b><i>Ctrl</i></b></kbd> and <kbd>Left</kbd>/<kbd>Right</kbd> -- Focus Workspace Left/Right

Moving:
- <kbd>Super</kbd>+<kbd><b><i>Shift</i></b></kbd> and <kbd>Left</kbd>/<kbd>Right</kbd> -- Move Current Window Left/Right
- <kbd>Super</kbd>+<kbd>Ctrl</kbd>+<kbd><b><i>Shift</i></b></kbd> and <kbd>Left</kbd>/<kbd>Right</kbd> -- Move Current Window to Next/Prev Workspace + Focus Next/Prev Workspace

Being able to use `Swipe: 3: Left` and `Swipe: 3: Right` instead of the left and right arrow keys would be awesome.

**Side Story**
I got this idea from a setup I have on my MacBook using BTT (BetterTouchTool). If I'm hovering over a window while holding <kbd>Ctrl</kbd>+<kbd>Opt</kbd> and I move my mouse, it will move the window under my cursor. Holding <kbd>Ctrl</kbd>+<kbd>Opt</kbd>+<kbd>Cmd</kbd> will resize the window instead. Paired with BTT's Window Snapping, it makes it super easy to Ctrl+Opt move a window to the top of the screen to maximize it. Pretty cool :)

iberianpig, 2022-06-29:
You can use the following plugins.
- https://github.com/iberianpig/fusuma-plugin-keypress
  - `keypress:` property while pressing with modifier keys
- https://github.com/iberianpig/fusuma-plugin-sendkey
  - `sendkey:` property for low-latency key event emulation

Following configuration is an example of changing behavior while modifier keys are pressed inGnome 42 on X11
```yaml
  4:
    left:
      sendkey: LEFTMETA+PAGEDOWN # move to next workspace
      keypress:
        LEFTSHIFT:
          sendkey: LEFTSHIFT+LEFTMETA+PAGEDOWN # move window to next workspace
    right:
      sendkey: LEFTMETA+PAGEUP # move to prev workspace
      keypress:
        LEFTSHIFT:
          sendkey: LEFTSHIFT+LEFTMETA+PAGEUP # move window to prev workspace
    up:
      sendkey: LEFTMETA+S
    down:
      sendkey: LEFTMETA+S
```

Swivelgames, 2022-06-29:
Awesome! Thank you! I suppose I could've done a bit more digging to find those haha.

closed 2022-06-29 by Swivelgames