iberianpig/fusuma #276: mentioning of AUR-Builds in the Archsection?

Shadowphoenix, 2022-07-09:
someone has built installscripts for the AUR, where you can pull all dependencies and all plugins as opt-depends - maybe mention them ^^

one would still need to add onesself to inputgroup but your command will do that

Link: https://aur.archlinux.org/packages/ruby-fusuma

iberianpig, 2022-09-09:
I think the install scripts are useful, but some issues make it difficult to let users know what is happening behind the scenes. In particular, if you add $USER to the input group, the executing user needs to know that /dev/inputs/ will be readable.

Shadowphoenix, 2022-09-13:
Maybe im just blissfully ignorant or blind but at least to me it seems this also does not get mentioned anywhere in your Readme/Documentations here on github.

Shadowphoenix, 2023-03-19:
yeap, checked again - /dev/input being readable when you are part of Group input is missing in your documentation here (so you probably want to edit your readme accordingly). will make a merge request about that now. merge at your own discretion.

Shadowphoenix, 2023-03-19:
https://github.com/iberianpig/fusuma/pull/289

closed 2023-03-19 by Shadowphoenix