iberianpig/fusuma #277: Fusuma setup instructions for Fedora

thaneesh-babu, 2022-08-31:
**Is your feature request related to a problem? Please describe.**
No instructions on setting up fusuma on Fedora.

**Describe the solution you'd like**
Update the README to include the following instructions:

#### 1. Install libinput-tools

You need `libinput` release 1.0 or later.

```bash
sudo dnf install libinput
```

#### 2. Install Ruby

Fusuma runs in Ruby, so you must install it first.

```bash
sudo dnf install ruby
```

#### 3. Install Fusuma

```bash
sudo dnf install fusuma
```

#### 4. Install xdotool (optional)

For sending shortcuts:

```bash
sudo dnf install xdotool
```

iberianpig, 2022-09-08:
Thanks for your suggestion.
Cloud you make a pull request for updating README.md?

TomSchammo, 2022-09-12:
I think there is actually a mistake here. Section 3 should be `sudo gem install fusuma`. Fusuma is not in the repos (at least not on Fedora 34).

thaneesh-babu, 2022-09-16:
> I think there is actually a mistake here. Section 3 should be `sudo gem install fusuma`. Fusuma is not in the repos (at least not on Fedora 34).

You're right, updating the issue now.

PR #278 (2022-09-17): Update README with Fusuma installation requirements on Fedora
Thank you for opening a Pull Request! Before submitting your PR, there are a few things you can do to make sure it goes smoothly:

- [ ] Make sure to open an issue as a [bug/issue](https://github.com/iberianpig/fusuma/issues) before writing your code!  That way we can discuss the change, evaluate designs, and agree on the general idea.

- [ ] Follow the instructions in [CONTRIBUTING](https://github.com/iberianpig/fusuma/blob/master/CONTRIBUTING.md). Most importantly, ensure the tests and linter pass by running `bundle exec rspec` an [... cut]
closed 2022-09-17 by iberianpig