iberianpig/fusuma #280: Feature request: Tip-Tap gesture

mauriciabad, 2022-11-30:
### Feature explanation

I'd like to use the Tip-Tap gestures (like the one in Better Touch Tool). 

It's a very simple gesture and very versatile. Let me highlight some of it's benefits:
- In contrast to the swipe, it has a it goes from 0% to 100% instantly, this provides a good UX for actions that have 2 counterparts and are instant, like copy/paste, prev/next song or any keyboard shortcut combination, whereas swipe is great for continuous actions like volume up/down, brightness up/down or movement-related actions.
- It can be performed consecutively very quickly. This is nice for actions that involve moving by steps, like prev/next window or tab.
- It's a natural, easy-to-perform and quick gesture. Making it great for frequent actions, like copy/paste.

Example useful actions that can be assigned:
- :star: Copy/Paste _(imho this is the ideal action, and should come enabled by default)_
- Shift + Tab / Tab keys
- Previous song or next song
- Go to prev/next Tab, Window, Workspace
- Tile window to left or right
- Increase/Decrease volume or brightness
- Remove/Delete keys
- Left/Right arrow keys (or Up/Down)

### What is Tip-Tap gesture?

1. Tap-hold with 1 finger
2. Tap with a 2nd finger
  a. Variant 1: Left
  b. Variant 2: Right

Video: https://www.youtube.com/watch?v=MRhBa3-O3fM

<p align="center">
  [image]
</p>

iberianpig, 2022-12-01:
Thank you for suggesting this interesting feature.
I am just not sure if libinput-debug-events can track the position of each tap with multiple fingers.
If it is possible, we can implement tip-tap as a fusuma-plugin.

smileBeda, 2023-11-18:
I am not entirely sure to follow.

The main request here was to implement a way to have the three-finger text highlight feature available in some way.
This is said to be possible now.

However:
1. With [fusuma-plugin-sendkey](https://github.com/iberianpig/fusuma-plugin-sendkey) you cannot do this, since it does not provide any _mouse_ commands - only keyboard commands are available
2. With xdotool you cannot implement this on a Wayland WM
3. With ydotool, you could probably implement this, but, you cannot implement window dragging, since yodtool does not have a mouse relative $variable position like xdotool has.

So I am a bit confused. It does not seem that it is possible to implement a three finger gesture to _highlight text_, still.
Implementing a drag (move window) or a swipe (like show "expose" etc) is not requiring any scripting: it can be done natively in gnome for the most part.

It is really the gestures like:
- three fingers on text used to highlight
- two fingers to navigate back (which I know is not possible at all anyway due to some "confusion" with scrolling, which while miyterious [mac does it...], is as is: not possible)
- three fingers move a window _when one does that swipe on a window header_

I can achieve the backwards navigation with three finger gesture, not a big deal, using either yodtool, native features or fusuma-plugin-sendkey. As well the drag-to-move window, either user native features or either of the ydotool or native features.

But the three-fingers-to-highlight, to me, seem still absolutely unachievable.

What do I miss?
Does ydotool have some hidden feature to grab mouse positions dynamically? Or fusuma a hidden tool to send _mouse_ commands?

Cheers!

mauriciabad, 2023-11-20:
@smileBeda It is not a 3 finger gesture, it is a 2 finger one. 

The difference between a "normal" 2 finger tap, is that both fingers don't touch the trackpad at the same time, instead, one goes first and some time after (for example 400ms), the other one comes in. Then, based the tap order of the fingers and their x-coordinate, the software triggers tip-tap left (if right finger 1st, left finger 2nd) or tip-tap right (if left 1sr, right 2nd)

That is the gesture part. And then, there's the action part. Here I'm just requesting the gesture, but obviously, if it comes with some actions it's better. In the GIF it rearranges windows, but that's a random GIF I found on the internet to show an example. What I would configure it to do is to trigger <kbd>Ctrl</kbd> + <kbd>C</kbd> and <kbd>Ctrl</kbd> + <kbd>V</kbd>, which seems trivial to me and I assumed it is already supported.

mauriciabad, 2023-11-20:
Now, I switched to macOS, and I use an app called [Better Touch Tool](https://folivora.ai/), which is the best gestures app by far (that I found for any OS). 

Take a look at the options it has, I recorded a video for you. It has almost all possible (and usable) gestures doable in a trackpad with up to 5 fingers. I also show some of the actions it supports.

https://github.com/iberianpig/fusuma/assets/12821361/1d5e7260-bc64-430f-b05b-380164690bb1

> Sadly Linux has a HORRIBLE trackpad usability, and that's my main way of interacting with the laptop, so I expect it to be perfect. I can't believe that the basic scrolling gesture is broken, it scrolls extremely fast and there's no option to adjust the sensitivity. The zoom gesture is also broken, and many things more... That's one of the reasons why I chose macOS. And now I won't be in constant pain. 
>
> I say this because I won't use this feature if you implement it, because I don't use linux anymore. But anyways, it would be a great adition, and the start of fixing the linux trackpad problem. (no library can do this kind of gestures in linux)

smileBeda, 2023-11-20:
... which is why I am so desperate for it on linux lol

Anyway, I got working ... default out of the box on X server. Forget wayland.
Also got most of the Mac gestures into it using this tool, and another (kinto)

But.. probably moving back to MacOS. Battery is a catastrophe on linux. So is half of the apps available. I guess it is the drawback of FOSS. Either pay, or pain 

Haha...

iberianpig, 2023-11-21:
I believe we need an event recognition and translation layer that does not rely on libinput, but rather uses evdev. This is necessary because libinput itself does not report touch positions. 

Currently, Fusuma operates based on libinput, but we could create plugins that utilize evdev events as needed.

For example, some functionalities of fusuma-plugin-thumbsense and fusuma-plugin-remap are implemented using evdev. This idea involves creating pairs of physical and virtual keyboards, and either rewriting the communication between them or using it as input for Fusuma.

Similarly, we can create pairs of physical and virtual touchpads, rewriting the communication between them. This approach would enable us to accurately detect touch positions and convert them into arbitrary events.

This means, by reporting touch positions while filtering the touches passed to the Wayland compositor's libinput, Fusuma could support more flexible gestures dependent on touch positions.