iberianpig/fusuma #281: Configurable hold gesture duration and/or long and short hold gesture

teackot, 2022-12-05:
**Is your feature request related to a problem? Please describe.**
Right now the hold gesture feels too long, I'd like to configure it.

**Describe the solution you'd like**
Custom duration in the config file. Also maybe different actions for different durations, i.e. long and short hold gestures.

**Describe alternatives you've considered**
Hardcode a lower duration for hold gestures.

iberianpig, 2022-12-06:
You can customize the duration of the hold gesture by using the threshold parameter. 

```yml
hold: 
  3:
    command: echo long press
    threshold: 2
  4:
    command: echo short press

threshold:
  hold: 0.2 #global 
```

Setting the threshold allows you to adjust the amount of time required for the hold gesture to be triggered. For example, setting a shorter threshold will cause the hold gesture to be triggered more quickly. Try adjusting the threshold parameter to achieve your desired hold gesture settings.

teackot, 2022-12-06:
I see, didn't have to use this parameter before, so I completely forgot about it. Thanks, closing the issue now

closed 2022-12-06 by teackot