iberianpig/fusuma #282: how can i use  ydotool replace xdotool

luyanfeng, 2023-01-04:
like this: 
`pinch:
  2:
    in:
      command: "xdotool keydown ctrl click 4 keyup ctrl" # threshold: 0.5, interval: 0.5
    out:
      command: "xdotool keydown ctrl click 5 keyup ctrl" # threshold: 0.5, interval: 0.5
`
how can i use  ydotool replace xdotool

iberianpig, 2023-01-04:
Please ask questions about individual commands in the ydotool repository.
It seems that ydotool has changed significantly since version 1.0.0, such as requiring a daemon, and is not a simple replacement for xdotool commands.

luyanfeng, 2023-01-04:
ok, thank you

> Please ask questions about ydotool in the ydotool repository. It seems that ydotool has changed significantly since version 1.0.0, such as requiring a daemon, and is not a simple replacement for xdotool commands.

closed 2023-01-04 by luyanfeng