iberianpig/fusuma #284: [Feature Request] Support lift up one finger and put down again when do three-fingers swipe

zyf0330, 2023-01-31:
**Is your feature request related to a problem? Please describe.**
I want that 3 fingers swipe feature is the same as on Mac.

**Describe the solution you'd like**
Currently, when do three-fingers swipe, after lift up one finder and put down again, swipe cannot be continud.

**Describe alternatives you've considered**
Maybe this is a technical problem.

**Additional context**

iberianpig, 2023-02-01:
Thank you for your feature request.

Are you referring to the following?

When a finger lifts off the touchpad for a split second during a 3-finger drag and becomes a two-finger drag, and then the finger immediately returns to a 3-finger drag.

At this time, the fusuma 3-finger drag is cancelled, but you want the 3-finger drag to continue?

zyf0330, 2023-02-01:
Yes, exactly right. I believe that you know the benefits of this feature, if you have used the 3-finger drag on the MacBook touchpad.

zyf0330, 2023-02-01:
And, I don't think the delay before returning to 3-finger drag must be **a split second**. I think it totally can be keeping 3-finger drag gesture except that all 3 finders lifts off the touchpad.

iberianpig, 2023-02-01:
Thanks for the details.

I need to research how Mac trackpad recognizes fingers when 3-finger dragging.
I'm going to go to an electronics store and study the behavior on Mac :smile: