iberianpig/fusuma #285: Three finger drag: "Can't make continuous selection/drag if lift the fingers for like one second"

vathanak-mao, 2023-02-21:
**Describe the bug**
It doesn't allow me to lift my fingers for a second then continue the selection. This must be allowed because sometimes we can't make a long drag or selection with a one go on a small trackpad.

**To Reproduce**
Steps to reproduce the behavior:
1. Start selecting a text with 3 fingers from left to right
2. Stop and left the fingers immediately for about one second
3. Start the selection again with 3 fingers from left to right

**Expected behavior**
The text selected from the first step and third step should be all selected. But, only the text from the third step is selected.

**Versions**
```yml
PASTE OUTPUT OF `$ fusuma --version`
```

**Configuration**
```yml
PASTE OUTPUT OF `$ cat ~/.config/fusuma/config.yml`
```

iberianpig, 2023-02-22:
Thanks for reporting.

Is this common behavior?
Is it possible to reselect previously selected characters after removing all fingers from the touchpad?

vathanak-mao, 2023-02-22:
Thank you iberianpig for replying.

I don't know whether it's commom on Windows but yes, it is on macOS cause I've been a Mac user for years.
Now, i'm using this program https://github.com/marsqing/libinput-three-finger-drag and it works flawlessly and it feels like i'm still using macOS. But, I like your project more and I don't want to use two programs at the same time.

Reselecting the previously selected characters (also redragging the prevously dragged window or file in nautilus) after removing all fingers from the touchpad should be possible only if the duration is like one second (or configurable). I hope you consider this the correct behavior too.

iberianpig, 2023-02-25:
I  want to make it possible to continue the drug as much as possible in future updates. Appreciate the good hint. Grateful for the feedback from users like you.

iberianpig, 2026-10-03:
This request duplicates the continuous drag behavior described in #320. I am closing this issue in favor of #320 so discussion and implementation ideas stay in one place. Please continue the discussion there.

closed 2026-10-03 by iberianpig