iberianpig/fusuma #286: Exception while pinching out to zoom out

tofylion, 2023-03-14:
**Describe the bug**
When using pinch out gesture with zoom out functionality, an exception happens and fusama crashes. This didn't happen all of the time, but it is happening consistently with me now. I'm not sure what caused it to start happening

**To Reproduce**
Steps to reproduce the behavior:
1. Create the config file like me
2. Try to pinch out

**Expected behavior**
Zoom out

**Versions**
```yml
I, [2023-03-14T16:23:57.070669 #27643]  INFO -- : reload config: /home/tofy/.config/fusuma/config.yml
I, [2023-03-14T16:23:57.096905 #27643]  INFO -- : ---------------------------------------------
I, [2023-03-14T16:23:57.097146 #27643]  INFO -- : Fusuma: 2.5.1
I, [2023-03-14T16:23:57.097882 #27643]  INFO -- : libinput: 1.20.0
I, [2023-03-14T16:23:57.097909 #27643]  INFO -- : ruby 3.0.2p107
I, [2023-03-14T16:23:57.098506 #27643]  INFO -- : OS: Linux 5.19.0-35-generic #36~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb 17 15:17:25 UTC 2
I, [2023-03-14T16:23:57.099102 #27643]  INFO -- : Distribution: elementary OS Horus \n \l
I, [2023-03-14T16:23:57.099722 #27643]  INFO -- : Desktop session: pantheon x11
I, [2023-03-14T16:23:57.099743 #27643]  INFO -- : ---------------------------------------------
I, [2023-03-14T16:23:57.099754 #27643]  INFO -- : Enabled Plugins: 
I, [2023-03-14T16:23:57.099818 #27643]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2023-03-14T16:23:57.099829 #27643]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2023-03-14T16:23:57.099837 #27643]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2023-03-14T16:23:57.099844 #27643]  INFO -- :   Fusuma::Plugin::Detectors::HoldDetector
I, [2023-03-14T16:23:57.099851 #27643]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2023-03-14T16:23:57.099858 #27643]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2023-03-14T16:23:57.099866 #27643]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2023-03-14T16:23:57.099874 #27643]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2023-03-14T16:23:57.099881 #27643]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2023-03-14T16:23:57.099888 #27643]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2023-03-14T16:23:57.099895 #27643]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2023-03-14T16:23:57.099901 #27643]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2023-03-14T16:23:57.099909 #27643]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2023-03-14T16:23:57.099916 #27643]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2023-03-14T16:23:57.099923 #27643]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2023-03-14T16:23:57.099929 #27643]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2023-03-14T16:23:57.099936 #27643]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2023-03-14T16:23:57.099942 #27643]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2023-03-14T16:23:57.099950 #27643]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2023-03-14T16:23:57.099957 #27643]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2023-03-14T16:23:57.099965 #27643]  INFO -- : ---------------------------------------------```

**Configuration**
```yml

swipe:
  4: #QUICK MEDIA CONTROL
    left:
      command: 'xdotool key alt+Right' # History forward
      keypress:
        LEFTCTRL: #Next Song
          command: 'xdotool key XF86AudioNext'
    
    right:
      command: 'xdotool key alt+Left' # History back
      keypress:
        LEFTCTRL: #Prev Song
          command: 'xdotool key XF86AudioPrev'

pinch:
  2:
    in:
      command: 'xdotool keydown Ctrl click 4 keyup Ctrl' # Zoom in
      threshold: 0.3
      interval: 0.2
    out:
      command: 'xdotool keydown Ctrl click 5 keyup Ctrl' # Zoom out
      threshold: 0.interval
      3: 0.2

rotate:
  3:
    clockwise:
      command: 'xdotool key XF86MonBrightnessUp' # brightness up
    counterclockwise:
      command: 'xdotool key XF86MonBrightnessDown' # brightness down
   
  4:
    clockwise:
      keypress:
        LEFTCTRL:
          command: 'xdotool key XF86AudioRaiseVolume' # Volume up
    counterclockwise:
      keypress:
        LEFTCTRL:
          command: 'xdotool key XF86AudioLowerVolume' # Volume down

      
# rotate:
#   2:
#     clockwise:
#       command: 'xdotool key XF86AudioRaiseVolume' # Volume up
#     counterclockwise:
#       command: 'xdotool key XF86AudioLowerVolume' # Volume down
#   3:
#     clockwise:
#       command: 'xdotool key XF86MonBrightnessUp' # Brightness up
#     counterclockwise:
#       command: 'xdotool key XF86MonBrightnessDown' # Brightness down

interval:
    rotate: 0.2

threshold:
    rotate: 0.5
plugin: 
  inputs:
    libinput_command_input: # options for lib/plugin/inputs/libinput_command_input
      enable-tap: true # click to tap
      enable-dwt: true # disable tap while typing
      show-keycodes: true # https://github.com/iberianpig/fusuma-plugin-keypress#add-show-keycode-option

```

iberianpig, 2023-03-14:
Your config.yml is broken.

```diff
@@ -17,18 +17,18 @@
 pinch:
   2:
     in:
       command: 'xdotool keydown Ctrl click 4 keyup Ctrl' # Zoom in
       threshold: 0.3
       interval: 0.2
     out:
       command: 'xdotool keydown Ctrl click 5 keyup Ctrl' # Zoom out
-      threshold: 0.interval
-      3: 0.2
+      threshold: 0.3
+      interval: 0.2

 rotate:
   3:
     clockwise:
       command: 'xdotool key XF86MonBrightnessUp' # brightness up
     counterclockwise:
       command: 'xdotool key XF86MonBrightnessDown' # brightness down
```

tofylion, 2023-03-14:
@iberianpig I just noticed that. That's kinda embarrassing 🙄. Thanks for the help!

closed 2023-03-14 by tofylion