iberianpig/fusuma #287: Two actions trigger on keypress when having window actions not on keypress

tofylion, 2023-03-14:
**Describe the bug**
When I'm trying to add 2 actions for the same gesture, one with a keypress and one without, they both activate with the keypress. This only happens with the window actions when they are set to be the ones without the keypress. In my config below, they are in the 4-finger swipe up and down gestures

**To Reproduce**
Steps to reproduce the behavior:
1. Set an action on keypress and another window action without the keypress
2. Try them out

**Expected behavior**
When clicking the key (ctrl), the action that's not on keypress shouldn't trigger. Only the action inside the keypress should trigger

**Versions**
```yml
I, [2023-03-14T19:22:25.581862 #38577]  INFO -- : reload config: /home/tofy/.config/fusuma/config.yml
I, [2023-03-14T19:22:25.618671 #38577]  INFO -- : ---------------------------------------------
I, [2023-03-14T19:22:25.619035 #38577]  INFO -- : Fusuma: 2.5.1
I, [2023-03-14T19:22:25.619828 #38577]  INFO -- : libinput: 1.20.0
I, [2023-03-14T19:22:25.619861 #38577]  INFO -- : ruby 3.0.2p107
I, [2023-03-14T19:22:25.620510 #38577]  INFO -- : OS: Linux 5.19.0-35-generic #36~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb 17 15:17:25 UTC 2
I, [2023-03-14T19:22:25.621035 #38577]  INFO -- : Distribution: elementary OS Horus \n \l
I, [2023-03-14T19:22:25.621489 #38577]  INFO -- : Desktop session: pantheon x11
I, [2023-03-14T19:22:25.621503 #38577]  INFO -- : ---------------------------------------------
I, [2023-03-14T19:22:25.621513 #38577]  INFO -- : Enabled Plugins: 
I, [2023-03-14T19:22:25.621584 #38577]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2023-03-14T19:22:25.621594 #38577]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2023-03-14T19:22:25.621602 #38577]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2023-03-14T19:22:25.621609 #38577]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2023-03-14T19:22:25.621616 #38577]  INFO -- :   Fusuma::Plugin::Detectors::HoldDetector
I, [2023-03-14T19:22:25.621623 #38577]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2023-03-14T19:22:25.621630 #38577]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2023-03-14T19:22:25.621638 #38577]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2023-03-14T19:22:25.621645 #38577]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2023-03-14T19:22:25.621652 #38577]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2023-03-14T19:22:25.621660 #38577]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2023-03-14T19:22:25.621667 #38577]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2023-03-14T19:22:25.621674 #38577]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2023-03-14T19:22:25.621682 #38577]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2023-03-14T19:22:25.621689 #38577]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2023-03-14T19:22:25.621696 #38577]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2023-03-14T19:22:25.621704 #38577]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2023-03-14T19:22:25.621712 #38577]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2023-03-14T19:22:25.621719 #38577]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2023-03-14T19:22:25.621726 #38577]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2023-03-14T19:22:25.621733 #38577]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2023-03-14T19:22:25.621740 #38577]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2023-03-14T19:22:25.621747 #38577]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2023-03-14T19:22:25.621755 #38577]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2023-03-14T19:22:25.621762 #38577]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2023-03-14T19:22:25.621769 #38577]  INFO -- : ---------------------------------------------
```

**Configuration**
```yml
swipe:
  4: #QUICK MEDIA CONTROL
    left:
      sendkey: "LEFTALT+RIGHT" # History forward
      keypress:
        LEFTCTRL: #Next Song
          sendkey: 'NEXTSONG'
    
    right:
      sendkey: "LEFTALT+LEFT" # History back
      keypress:
        LEFTCTRL: #Prev Song
          sendkey: 'PREVIOUSSONG'
    up:
      window:
        maximized: 'add'
      keypress:
        LEFTCTRL:
          sendkey: 'LEFTCTRL+T'
    down:
      keypress:
        LEFTCTRL:
          sendkey: 'LEFTCTRL+W'
      window:
        maximized: remove

hold:
  4:
    sendkey: PLAYPAUSE
  3:
    sendkey: 'LEFTCTRL+LEFTSHIFT+T'

pinch:
  2:
    in:
      # sendkey: 'LEFTCTRL+KPPLUS'
      command: 'xdotool keydown Ctrl click 4 keyup Ctrl' # Zoom in
      threshold: 0.3
      interval: 0.2
    out:
      # sendkey: 'ZOOMOUT'
      command: 'xdotool keydown Ctrl click 5 keyup Ctrl' # Zoom out
      threshold: 0.3
      interval: 0.2

rotate:
  4:
    clockwise:
      sendkey: 'BRIGHTNESSUP'
      keypress:
        LEFTCTRL:
          command: 'xdotool key XF86AudioRaiseVolume' # Volume up
    counterclockwise:
      sendkey: 'BRIGHTNESSDOWN'
      keypress:
        LEFTCTRL:
          command: 'xdotool key XF86AudioLowerVolume' # Volume down

interval:
    rotate: 0.2

threshold:
    rotate: 0.5
plugin: 
  inputs:
    libinput_command_input: # options for lib/plugin/inputs/libinput_command_input
      enable-tap: true # click to tap
      enable-dwt: true # disable tap while typing
      show-keycodes: true # https://github.com/iberianpig/fusuma-plugin-keypress#add-show-keycode-option
      verbose: true
```

iberianpig, 2023-03-17:
@tofylion  Have you assigned other keys(e,g CAPSLOCK) to LEFTCTRL?
I use an external tool called xremap to change keybind from CAPSLOCK to LEFTCTRL, and in my environment, the bug you reported seems to be reproduced.

Actually, when Fusuma executes sendkey, it intentionally releases the modifier keys other than the key specified by keypress before executing sendkey. This includes CAPSLOCK, and if Fusuma releases CAPSLOCK, the remapped LEFTCTRL will also be released, which may cause this kind of bug.

Below is a list of modifier keys that are automatically released just before sendkey.
```
          CAPSLOCK
          LEFTALT
          LEFTCTRL
          LEFTMETA
          LEFTSHIFT
          RIGHTALT
          RIGHTCTRL
          RIGHTSHIFT
          RIGHTMETA
```

iberianpig, 2023-03-17:
What I'm thinking of now is to make the automatic release of modifier keys an option.
This may not work properly in some cases, but I think it's better than being released automatically. I think it's best to minimize the surprise.

closed 2025-07-08 by iberianpig