iberianpig/fusuma #291: Pinch using 4 fingers not working on eOS at all

OmarShawky1, 2023-03-19:
**Describe the bug**
Pinch not working on elementaryOS 6.0

**Expected behavior**
Pinch in using 4 fingers should press "Alt_L + F4" to close window. That doesn't happen although fusuma detects 4 fingers successfully.

`D, [2023-03-19T16:27:16.296841 #14472] DEBUG -- : {:input_event=>time: 2023-03-19 16:27:16 +0200, tag: libinput_command_input, record:  event5   GESTURE_PINCH_END +15.661s	4}`

**Versions**
```yml
$ fusuma --version
I, [2023-03-19T16:25:13.100325 #14308]  INFO -- : reload config: /home/omar/.config/fusuma/config.yml
I, [2023-03-19T16:25:13.121445 #14308]  INFO -- : ---------------------------------------------
I, [2023-03-19T16:25:13.121645 #14308]  INFO -- : Fusuma: 2.5.1
I, [2023-03-19T16:25:13.123110 #14308]  INFO -- : libinput: 1.15.5
I, [2023-03-19T16:25:13.123132 #14308]  INFO -- : ruby 2.7.0p0
I, [2023-03-19T16:25:13.123820 #14308]  INFO -- : OS: Linux 5.15.0-67-generic #74~20.04.1-Ubuntu SMP Wed Feb 22 14:52:34 UTC 2023
I, [2023-03-19T16:25:13.124422 #14308]  INFO -- : Distribution: elementary OS Jólnir \n \l
I, [2023-03-19T16:25:13.124908 #14308]  INFO -- : Desktop session: pantheon x11
I, [2023-03-19T16:25:13.124928 #14308]  INFO -- : ---------------------------------------------
I, [2023-03-19T16:25:13.124937 #14308]  INFO -- : Enabled Plugins: 
I, [2023-03-19T16:25:13.124983 #14308]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2023-03-19T16:25:13.124993 #14308]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2023-03-19T16:25:13.125000 #14308]  INFO -- :   Fusuma::Plugin::Detectors::HoldDetector
I, [2023-03-19T16:25:13.125007 #14308]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2023-03-19T16:25:13.125014 #14308]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2023-03-19T16:25:13.125021 #14308]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2023-03-19T16:25:13.125027 #14308]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2023-03-19T16:25:13.125034 #14308]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2023-03-19T16:25:13.125040 #14308]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2023-03-19T16:25:13.125047 #14308]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2023-03-19T16:25:13.125054 #14308]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2023-03-19T16:25:13.125061 #14308]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2023-03-19T16:25:13.125067 #14308]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2023-03-19T16:25:13.125074 #14308]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2023-03-19T16:25:13.125097 #14308]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2023-03-19T16:25:13.125104 #14308]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2023-03-19T16:25:13.125111 #14308]  INFO -- : ---------------------------------------------
```

**Configuration**
```yml
swipe:
        3:
                right:
                        command: "xdotool key 0xffe9+0xff09"
                left:
                        command: "xdotool key 0xffe9+0xff51"
        4:
                up:
                        command: "xdotool key super+0xff52"
                down:
                        command: "xdotool key super+0xff54"
pinch:
        4:
                in:
                        command: "xdotool key 0xffe9+0xffc1"
```

iberianpig, 2023-03-25:
Thank you for reporting the issue.
I also feel that the behavior of the 4-finger pinch is a bit strange on my end. I think it's probably an issue with the internal threshold calculation. I confirmed that it triggers with 3 fingers. Does the pinch operation work if you configure it as a 3-finger pinch?

iberianpig, 2023-03-25:
Please try lowering threshold in 4-finger pinch. It works around 0.5 in my environment(Ubuntu 22.04/x11/XPS 9130).
Also, it seems that fusuma recognizes pinch in and pinch out in reverse :sweat_smile: 

```yaml
pinch:
        4:
                in:
                        command: "xdotool key 0xffe9+0xffc1"
                        threshold: 0.5
```

iberianpig, 2023-03-26:
The pinch direction reversing operation was fixed in https://github.com/iberianpig/fusuma/commit/aa46ae1a86efcbc265c3a43c4c73fba445b4ca81

OmarShawky1, 2023-03-26:
> Please try lowering threshold in 4-finger pinch. It works around 0.5 in my environment(Ubuntu 22.04/x11/XPS 9130). Also, it seems that fusuma recognizes pinch in and pinch out in reverse [...]

Yes, It worked when I used the threshold you requested, thank you so much.

Yet, I don't understand why?

iberianpig, 2023-03-26:
The base value of the threshold for pinch gesture should be changed according to the number of fingers, maybe.
The reason is because the distance of finger movement is shorter for 4-finger pinch gesture than 2-finger pinch gesture.

closed 2023-03-27 by OmarShawky1