iberianpig/fusuma #292: [Question] Preventing default gesture behavior

hyunsooda, 2023-04-11:
I am currently using Ubuntu 22.04 LTS as my host environment, which comes with a default touchpad gesture that takes up some of the motions. Specifically, I require a new configuration for the swipe motion using three fingers, but the default configuration already has a predefined behavior for this motion. As a result, both behaviors are triggered simultaneously, causing annoyance and eventually preventing me from using this motion altogether.

I understand that this may be beyond the scope of this repository, but I would appreciate any advice from someone who has successfully resolved the same issue. Thank you.

iberianpig, 2023-04-13:
For example, you can change the 3-finger gesture to a 4-finger gesture. It's not the best way, but you can modify the code in libgnome-shell.so with a js code.
https://gist.github.com/iberianpig/6e7049a02489950f570b4a27cff8df5e

It seems that you can change the default gesture using a Gnome extension, but I am not familiar with Gnome extensions.

hyunsooda, 2023-04-17:
Hello @iberianpig,

I have tested some GNOME extensions, but they only offer one setting that changes the 3-finger gesture to a 4-finger gesture. What I would like to do is modify a different behavior, such as the right 3-finger swipe to trigger `ctrl+tab`.

Thank you.

iberianpig, 2023-05-01:
How about trying to change the number of fingers for the default swipe in wayland to 4 using:

```sh
sudo sed -e '/const GESTURE_FINGER_COUNT = [[:digit:]];/ s/[[:digit:]]/4/' /usr/lib/gnome-shell/libgnome-shell.so -i
```
and setting the 3-finger swipe in fusuma?

hyunsooda, 2023-05-01:
AMAZING!!! This is exactly what I had been hoping for. It's completely resolved. Thank you so much! I was struggling to find this source before. May I ask where you obtained it?"

iberianpig, 2023-05-01:
I think maybe this https://www.reddit.com/r/gnome/comments/mndocf/is_there_any_way_to_switch_back_to_4finger/

hyunsooda, 2023-05-02:
Thank you! Let me close this.