iberianpig/fusuma #293: [Question]is it usable for touchscreen?

lVentus, 2023-04-28:
Hello! I wonder if it can work for touchscreen? I have a surface pro, and I want it to work with touchscreen as well, when it is in tablet mode. Is it possible? Thanks a lot!

iberianpig, 2023-04-29:
Fusuma does not handle touchscreen events, so it should not work.
It is possible to implement the feature improvement, but it is difficult because I do not have a touchscreen device.

Phaengris, 2023-05-19:
Don't want to give you too much hope till the work is done, but at the moment I'm working on a plugin to give Fusuma such ability. By the very same reason - I have surface pro and I want Fusuma to work on it :)

Phaengris, 2023-05-19:
Don't want to give you too much hope till the work is done, but at the moment I'm working on a plugin which should allow Fusuma to work on touchscreens. By the very same reason - I have a surface pro and I want Fusuma to work on it :)

iberianpig, 2023-05-20:
That's great! Feel free to ask any questions you have. I'm here to help!

Phaengris, 2023-06-18:
Some progress report :) sorry for being slow with updates, I have not much time to work on this...
- added swipe support
- heavily reworked the plugin's structure

The next steps: add hold support, then release 0.0.1, then work on pinch / rotate / edge.
(Or add pinch / rotate, then release 0.0.1? :thinking:)

iberianpig, 2023-06-20:
That's really amazing!
I'm thinking of buying a touchscreen device myself so that I can try it out :smile: 

I think it's a good idea to release the features gradually, in smaller increments.

Phaengris, 2023-06-20:
Progress report: hold support.

@iberianpig you're probably right :+1: let me release it and then add features.

Phaengris, 2023-06-21:
Well, I've released it as a gem https://rubygems.org/gems/fusuma-plugin-touchscreen/versions/0.0.1.alpha

I still consider it an alpha version :) but releasing it as a gem may give a chance to install it for everyone who would like to try.

I need to write some readme at least and also add some tests to make it non-alpha.

Phaengris, 2023-07-10:
I was away from development for a time, but now there's some news:
- pinch implemented
- rotate implemented
What's left to do is to finish a spec for TouchBuffer, then release 0.1.0

Phaengris, 2023-07-17:
Well, 0.1.0 is released, readme and changelog are updated. So now we can say, with some caution :) that Fusuma has support for touchscreens :)

Now I'll start to use it on my device and see if there are any bugs to fix / opportunities to improve. Anyone who can / want to test it is very welcome!

https://github.com/Phaengris/fusuma-plugin-touchscreen