iberianpig/fusuma #295: How to make main pipeline more efficient?

ces42, 2023-05-29:
**Is your feature request related to a problem? Please describe.**
Fusuma has a somewhat high cpu usage (~9% while performing a swipe on my laptop) for a background process performing a task that isn't computationally intensive.
I was looking at the "main pipeline" to see if there are any inefficiencies in the way events are processed. There seem to be a few. 

Running fusuma with the builtin profiles shows that lines 21 and 22 in [`swipe_detector.rb`](https://github.com/iberianpig/fusuma/blob/master/lib/fusuma/plugin/detectors/swipe_detector.rb)  take a significant amount of time https://github.com/iberianpig/fusuma/blob/b68ab60335260dc32bfb3748c0bf65f9dc6d9dad/lib/fusuma/plugin/detectors/swipe_detector.rb#L20-L22
Closer insepction of the function definitions in [`gesture_buffer.rb`](https://github.com/iberianpig/fusuma/blob/master/lib/fusuma/plugin/buffers/gesture_buffer.rb) shows that this is due to searching long arrays that contain all events since the beginning of the last gesture in lines 102 https://github.com/iberianpig/fusuma/blob/b68ab60335260dc32bfb3748c0bf65f9dc6d9dad/lib/fusuma/plugin/buffers/gesture_buffer.rb#L102 and 66 https://github.com/iberianpig/fusuma/blob/b68ab60335260dc32bfb3748c0bf65f9dc6d9dad/lib/fusuma/plugin/buffers/gesture_buffer.rb#L66

Since this search is performed at every event, the work that is done is **quadratic** in the number of events in one gesture, and touchpads emit more than 100 events per second.

It seems like the "merge" step of the pipeline also takes plenty of time, but I don't really understand the code there well enough to identify the reason.

iberianpig, 2023-05-30:
Thank you for creating the issue and the pull request. I'm really happy about it.
I'm currently unable to dedicate time immediately, so it may take some time to review and reply. :pray:

ces42, 2023-05-30:
> it may take some time to review and reply.

No worries!

I did some more changes to the pr and I've been trying to find a good way to reliably measure the speedups. I might make some more changed in the coming days, so no need to rush. :)