iberianpig/fusuma #305: Can fusuma Execute before or after gesture actions

luyanfeng, 2023-08-10:
Can fusuma Execute before or after gesture actions ?

And Can you add some animations ？

iberianpig, 2023-08-12:
@luyanfeng 

Currently, fusuma does not support executing actions before or after gestures. However, depending on the use case, you might be able to assign commands at the beginning and end using the `begin`, `update`, and `end` properties.

Regarding adding animations, we do not want to support it as it would heavily depend on specific distributions or desktop environments. However, if there's an animation API available, I believe fusuma plugin could utilize it.

Thank you for your suggestion.

luyanfeng, 2023-08-14:
I like fusuma.  It`s too helpfully for me.

closed 2023-08-14 by luyanfeng