iberianpig/fusuma #307: Can you add a new parameter to let me execute each gesture only once

luyanfeng, 2023-08-14:
Can you add a new parameter to let me execute each gesture only once e. g: 

For example, this gesture is used to close a app window：
(I just want to trigger this only once every time I execute this gesture)
```yaml
pinch:
  3:
    out: 
      sendkey: "LEFTALT+F4"  
      threshold: 0.5
      interval: 0.5
      times: 1  # new parameter : this gesture triggers only once.
```
But now, it sometimes closes multiple overlapping windows, This is too scary !

iberianpig, 2023-08-14:
Setting large interval like interval: 5,  doesn't fit your usecase?

luyanfeng, 2023-08-18:
> Setting large interval like interval: 5, doesn't fit your usecase?

yes, it works well.

closed 2023-08-18 by luyanfeng