iberianpig/fusuma #308: browser forward and backward

luyanfeng, 2023-08-29:
Can it support two finger operations, such as browser forward and backward

iberianpig, 2023-08-29:
There are no plans to support it in fusuma.
It should be implemented on browser side.
I think that implementing this naively would make it is not useful. For example, if it hijacked horizontal scrolling, it will not possible to scroll on whiteboard tools in browser. Is it good user interface?

IceAsteroid, 2023-09-15:
@luyanfeng At least, in firefox and under wayland, it has been able to toggle backward/forward by two finger swiping gestures natively, by default.

I also tested chromium, but it seemed it didn't work by default.

You can have gestures bound to specific commands with a key simulation tool like xdotool to simulate the key presses as, for example, "alt+left_arrow" and "alt+right_arrow" for chromium 

The following keys to specify with xdotool are just examples, since, I'm using wtype under wayland

luyanfeng, 2023-09-22:
> @luyanfeng In firefox and under wayland, it has been able to toggle backward/forward by two finger swiping gestures natively, by default.

I don't know what the problem is, but I tried it under Wayland and found that both Firefox and Chrome were not working

smileBeda, 2023-11-30:
This is possible using `syngestures`, see https://github.com/mqudsi/syngesture
You can use this config:
```
[[device]]
device = "/dev/input/by-path/pci-0000:04:00.1-usb-0:5:1.2-event-mouse"#ADJUST AS NEEDED FOR YOUR CASE
gestures = [
	# Navigate next
	{ type = "swipe", direction = "right", fingers = 2, execute = "xdotool key alt+Left" },
	# Navigate previous
	{ type = "swipe", direction = "left", fingers = 2, execute = "xdotool key alt+Right" },
]
```

It does not hijack anything, since a navigation is a _fast_ swipe and a scroll is a _slow movement_. Entirely different things on a proper clickpad.

The above works in _most_ windows (surely in all browsers), including nautilus, discord app, etc.

iberianpig, 2023-12-05:
Thank you, smileBeda, for suggesting the use of syngestures. 
I'm planning to try it out for the two-finger gestures for browser forward and backward navigation. Based on how well it works, I'll reconsider implementing this feature in Fusuma.
Your suggestion seems very helpful. I might share feedback on the user experience after trying it out. Thanks again for your support.

closed 2023-12-05 by luyanfeng

smileBeda, 2023-12-05:
@luyanfeng - unfortunately it is wonky. The more I used, the less I was happy about it, and had to disable it.
There is no treshold! Meaning, you barely touch the clickpad in a horizontal swipe fashion (literally barely, you even wonder how it is horizontal or swipe at all, so minimal it is) and the browser navigates back/forth.

Great, when you test it, you think "wow it is snappy". Awful, infuriating, when you use it in daily usage.

---

I really do not feel clickpad can be made anywhere close to macOS's experience.
1. Given all commands are under the hood actual _key strokes_, things are way too "blocky"
2. The intermittent "hangs" do not contribute to a better experience.

As an example, if you switch workspaces on a mac, you can actually determine how fast a workspace swicthes by the speed of your drag. You can even hold it there, "in the between", and "peek" at contents of one workspace and the other at the same time (half/half). Given on linux we have to translate gesture to keystrokes, the workspace switch is immediate, uncontrollable so to say.

The same translates into any other gesture (apart of maybe the "drag window around" gesture).

This is really unfortunate, and I would wish there where another way to implement a MacOS like gesture on linux. However I cannot find any other solution, and I have no idea how MacOS does it (it must be "copyable", said code, it cannot be some black magic, but... I just do not know how they do it lol)

That said, this fusuma here is at least approaching a likeness and massively improves the absolutely awful interpretation of gestures "behaviour" that gnome or linux in general believes to be best for a clickpad!
Thanks for that for sure :)

luyanfeng, 2023-12-06:
> @luyanfeng \ @iberianpig - unfortunately it is wonky. The more I used, the less I was happy about it, and had to disable it. There is no treshold! Meaning, you barely touch the clickpad in a horizontal swipe fashion (literally barely, you even wonder how it is horizontal or swipe at all, so minimal it is) and the browser navigates back/forth.
> 
> Great, when you test it, you think "wow it is snappy". Awful, infuriating, when you use it in daily usage.
> 
> So I would LOVE to see this in fusuma, because fusuma has treshold!!!
> 
> I really do not feel clickpad can be made anywhere close to macOS's experience (which if not already apparent, is my actual goal)
> 
> 1. Given all commands are under the hood actual _key strokes_, things are way too "blocky"
> 2. The intermittent "hangs" do not contribute to a better experience.
> 
> As an example, if you switch workspaces on a mac, you can actually determine how fast a workspace swicthes by the speed of your drag. You can even hold it there, "in the between", and "peek" at contents of one workspace and the other at the same time (half/half). Given on linux we have to translate gesture to keystrokes, the workspace switch is immediate, uncontrollable so to say.
> 
> The same translates into any other gesture (apart of maybe the "drag window around" gesture).
> 
> This is really unfortunate, and I would wish there where another way to implement a MacOS like gesture on linux. However I cannot find any other solution, and I have no idea how MacOS does it (it must be "copyable", said code, it cannot be some black magic, but... I just do not know how they do it lol)
> 
> That said, this fusuma here is at least approaching a likeness and massively improves the absolutely awful interpretation of gestures "behaviour" that gnome or linux in general believes to be best for a clickpad! Thanks for that for sure :)
> 
> However, if you think there is _any_ chance to achieve a more smooth experience - I am ready to help, with whatever I can. I am proficient in Python (and PHP, which here is not really useful lol), so if I can help in any way... let me know. I also own a mac and if needed I am ready to dissect their code - but I do not even know where to start about that, macos is a blackbox to me. I also own a company so I am ready to help financially too, to some extent. Really if we could get these clickpads to work on linux, I believe it would be a major kick.

You are right. I found , In some scenarios that require horizontal sliding, [syngeture](https://github.com/mqudsi/syngeture) cannot stop, and it cannot identify when it needs to be executed and when it does not. Finally I also disabled it