iberianpig/fusuma #310: [Feature request] Slant gestures

IceAsteroid, 2023-09-12:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
Slant gestures such as:
	Swipe from left to right but slantly up/down
	Swipe from right to left but slantly up/down

**Why I post this issue and not just libinput-gestures instead, since libinput-gestures supports this feature**
The developer of libinput-gestures has said that project is only as a stop-gap measure before things work natively.

And the most important thing is, fusuma conflicts with libinput-gestures, as the former interprets those slant gestures as perpendicular.

For example, if I set a slant left gesture for firefox in libinput-gesture, and a perpendicular left gesture for workspace in fusuma, and I make a slant left gesture on touchpad, they'll all be triggered at the same time. 

**Why this is necessary**
Two finger swipe gestures are already used for scrolling.

Three finger swipe gestures are often used for workspace switching.

Four swipe gestures are often used for calling system utilities such as application launcher and workspace overview panel.

It would be really handy if slant swipe gestures could be added for navigation in a browse, for instance.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context about the feature request here.

iberianpig, 2023-09-12:
Thanks for your feature request.

While the implementation itself might not be difficult, there could be an increased number of false detections for the existing up, down, left, and right swipes on smaller touchpads.

I'll offer another idea. 
Make the most out of fusuma's plugins.

- Set the 3-finger gestures for each application.
    - Use fusuma-plugin-appmatcher to configure settings for each application.
- Assign the 4-finger gestures to workspace or window manager level operations.
    - Use the default swipe without modifier keys for workspace movement, and with fusuma-plugin-keypress, assign window operations and other utilities while holding Shift or Super.

This is how I do it. It's very convenient :pig:

IceAsteroid, 2023-09-12:
@iberianpig Hi, thank you for the kind reply and patience! :)

I would like to say yes, but actually, I asked this for a reason..

My swaywm config is fairly complex but yet efficient(opinionated), where I can have more than 10 workspaces, or any number of them set by set..

Not just to switch between them back and forth, I also need a pair of gestures to switch between "digits", for example, 12th workspace to 22th or N2th workspace.

In my prescription, ones of the four finger gestures are to switch between these digits.. 

It's probably tricky for a small touchpad to use slant swipe gestures, but is it feasible to be enabled when people need it?  

**About using app-specific gestures/with modifiers**
I also have a badass script to disable cursor movements when type and the cursor needs to be enabled by the win key. So if it needs to move a cursor out of the app, then to use to the same gesture for other purposes, and it would be hassle enough.

https://github.com/iberianpig/fusuma/assets/90975914/22e2c06c-02da-4913-9881-74d456ced4f1

**Conclusion**
Primarily, it's just my own problem, but here I just let you guys know the cases of why sometimes slant swiping gestures are useful for, though..

IceAsteroid, 2023-09-27:
Any good news?

iberianpig, 2023-10-09:
I am not highly motivated to implement diagonal swipe gestures. This is because I am concerned about false positives in the case of Fusuma, which defaults to thresholds. If you need to, please try to create a pull request for PoC.

IceAsteroid, 2023-10-11:
Fine, I respect it and your choice. I'll try to create it.

But it may take for months or a year, since I'm in trouble in reality and still learning python as my first proper programming language after my bash skills.

Which means I also don't know how to program in Ruby, but it's ok, if that's the case only for me to require slant swiping gestures.

That's it from me, sincerely :)