iberianpig/fusuma #314: Can't autostart fusuma on fedora because was removed a full ten years ago.

orcbolg3612, 2023-12-24:
How can i autostart fusuma on fedora.

orcbolg3612, 2023-12-24:
╭─orcbolg@fedora ~
╰─$ gnome-session-properties
zsh: gnome-session-properties: command not found…

I installed gnome-session and i am using fedora39 with latest update.
╰─$ sudo dnf install gnome-session 127 ↵
[sudo] password for orcbolg:
Last metadata expiration check: 0:50:40 ago on Sun 24 Dec 2023 08:55:52 AM +07.
Package gnome-session-45.0-3.fc39.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!

iberianpig, 2024-07-14:
I think that you can use `~/.config/autostart/fusuma.desktop` to set up autostart.

Try creating the following .desktop file:

```ini
[Desktop Entry]
Name=fusuma
Comment=run fusuma
# If it doesn't work, check the path with `which fusuma`
Exec=fusuma -d --log=/tmp/fusuma.log --verbose
Icon=input-touchpad
X-GNOME-Autostart-enabled=true
Type=Application
```

After creating the file, make sure its permissions are correct, and then check whether it starts automatically. Especially ensure the `Exec` path is correct by running the `which fusuma` command in the terminal to get the correct path and use it instead of `fusuma` in the `Exec` line if necessary.

iberianpig, 2025-01-26:
I've updated the README to include instructions for manually creating a desktop entry to autostart Fusuma. This provides an alternative method in case `gnome-session-properties` is not available on like your Fedora setup. Please refer to the "Autostart" section in the README for the detailed steps.

If you encounter any issues, feel free to reach out. Thank you for bringing this to our attention!

closed 2025-01-26 by iberianpig