iberianpig/fusuma #320: Tap & Drag (Retain Tap Threshold When Finger Lifted)

jncsotk30, 2024-06-10:
**Is your feature request related to a problem? Please describe.**
This isn't really a problem, but I'm trying to determine if this already exists or can be implemented.

**Describe the solution you'd like**
While using three finger drag or tap and drag, I'd like to be able to lift my fingers off the track-pad for a brief second to reposition my fingers back at other end of track-pad and continue dragging. I'd like there to be some kind of tap retention ability so we can reposition fingers on track-pad to continue the drag operation. 

**Describe alternatives you've considered**
I haven't found any alternative options to this functionality.

**Additional context**
An example use case of this would be screenshot of area while using track-pad.

1. Initialize screenshot tool for (drag selection of area)
2. Place finger(s) at top left of selection
3. Drag down and to right of screenshot area
4. Lift fingers and reposition them on other side of track-pad (this is where functionality would be needed / retain the tap and drag operation)
5. Continue dragging to complete the selection area for screenshot
6. Once fingers lifted for threshold amount, tap and drag ends.

iberianpig, 2024-06-11:
Thanks for your interesting suggestion!

Implementing the continuous three-finger drag feature seems a bit challenging at the moment. We're not sure if or when we can roll it out, but we've definitely taken note of it as a feature request.

In the meantime, though not a complete alternative, you might find the Thumbsense extension really useful for drag-and-drop actions. Personally, I found that after using Thumbsense, I didn't need the three-finger drag feature anymore. Give it a try!

I've posted a GIF showcasing how it works in the link below:
https://dev.to/iberianpig/enhance-your-touchpad-experience-on-linux-with-thumbsense-391n

You can also check out the repository here: https://github.com/iberianpig/fusuma-plugin-thumbsense

jncsotk30, 2024-06-22:
Hi, thanks also for the suggestion and reply. Hope to see this considered in the future. :)

Ah, I've never heard of ThumbSense. Interesting project. So, I could use the thumb sense lib to retain the tap and hold functionality while dragging something around. I may give it a try. Having my thumb on the trackpad at the same time as dragging might be a bit cramped tbh.

Thanks againi!

iberianpig, 2024-06-22:
This feature is the second invention I brought to Linux. The first was xSwipe, which came before Fusuma, and the second is Thumbsense. Using your index finger, not just your thumb, for pointing is also very useful. It makes clicking and dragging much smoother.

Please, you must try it. It's amazing!