iberianpig/fusuma #321: package installed on root instead of personal user

Guyc1800, 2024-06-30:
openSUSE Tumbleweed x86_64
Linux 6.9.6-1-default
KDE Plasma 6.1.1 , Wayland
gem version=3.5.11
**the bug**
when running `sudo gem install fusuma` the config file is created on root/.config/fusuma instead of /home/<user>/.config/fusuma and fusuama command avaliable only via root user

iberianpig, 2024-06-30:
The gem install fusuma command does not create configuration files. Are you using sudo su during the installation? This can cause the configuration files to be placed in the root directory instead of the user's home directory.

Guyc1800, 2024-06-30:
I am using sudo

iberianpig, 2024-06-30:
It seems that there might be a mistake. Perhaps you accidentally executed the following commands with `sudo`:

```
sudo mkdir -p ~/.config/fusuma
sudo nano ~/.config/fusuma/config.yml
```

This could cause the configuration files to be placed in the root directory instead of the user's home directory. You need to create the config files without using `sudo`. For more details, please refer to the [README](https://github.com/iberianpig/fusuma?tab=readme-ov-file#customize-gesture-mapping).

Guyc1800, 2024-06-30:
As I said before the only thing I did is run `sudo gem install fusuma` exactly as the README says, there is no point to argue about it but if you need any further information about the bug I am willing to share

Guyc1800, 2024-07-02:
update: i just tried to install the package again and for some reason now it works as expected. closeing the issue

closed 2024-07-02 by Guyc1800