iberianpig/fusuma #322: Default values for fuguma

Shantanu2k19, 2024-08-23:
I have created a configuration file for Fusuma with the following settings:

```yaml
swipe:
  3:
    left:
      command: 'xdotool key alt+Left'  # Back in browser
    right:
      command: 'xdotool key alt+Right' # Forward in browser
```
configuration file at /home/zodiac/.config/fusuma/config.yml, and it is the only configuration file present in that directory.

Despite only defining actions for two gestures (three-finger swipe left and right), Fusuma is also performing actions for other gestures, such as three-finger swipe down and swipe up.

**Question**: Where is Fusuma sourcing commands for these additional gestures? Is there a default configuration file that Fusuma is using for gestures that I haven't explicitly defined?

Shantanu2k19, 2024-08-23:
using ubuntu 24.04 

attaching debug logs : 
```pyhton
   +35.087s	2 cancelled}
D, [2024-08-23T23:25:23.950314 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_BEGIN      +37.724s	2}
D, [2024-08-23T23:25:23.972832 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_END        +37.746s	2 cancelled}
D, [2024-08-23T23:25:24.689187 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_BEGIN      +38.462s	2}
D, [2024-08-23T23:25:24.703809 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_END        +38.477s	2 cancelled}
D, [2024-08-23T23:25:29.128236 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_BEGIN      +42.902s	2}
D, [2024-08-23T23:25:29.136273 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_END        +42.910s	2 cancelled}
D, [2024-08-23T23:25:29.445983 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_BEGIN      +43.219s	2}
D, [2024-08-23T23:25:29.447073 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_END        +43.221s	2 cancelled}
D, [2024-08-23T23:25:30.308473 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_BEGIN      +44.082s	1}
D, [2024-08-23T23:25:30.337183 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_END        +44.110s	1}
D, [2024-08-23T23:25:31.288413 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_BEGIN      +45.062s	1}
D, [2024-08-23T23:25:31.310755 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_END        +45.084s	1}
D, [2024-08-23T23:25:32.017205 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_BEGIN     +45.791s	3}
D, [2024-08-23T23:25:32.017606 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.791s	3 -13.46/ 1.21 (-13.46/ 1.21 unaccelerated)}
D, [2024-08-23T23:25:32.025010 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.798s	3 -15.48/ 1.36 (-14.93/ 1.32 unaccelerated)}
I, [2024-08-23T23:25:32.025787 #149594]  INFO -- : {:command=>"xdotool key alt+Right", :args=>{:move_x=>-15.48, :move_y=>1.36, :unaccelerated_x=>-14.93, :unaccelerated_y=>1.32, :zoom=>0.0, :rotate=>0.0}}
D, [2024-08-23T23:25:32.032713 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.806s	3 -19.84/ 1.65 (-15.84/ 1.32 unaccelerated)}
D, [2024-08-23T23:25:32.040143 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.813s	3 -21.82/ 1.59 (-16.52/ 1.21 unaccelerated)}
D, [2024-08-23T23:25:32.047895 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.821s	3 -22.90/ 1.79 (-16.86/ 1.32 unaccelerated)}
D, [2024-08-23T23:25:32.055255 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.829s	3 -24.56/ 2.30 (-17.54/ 1.64 unaccelerated)}
D, [2024-08-23T23:25:32.062673 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.836s	3 -27.16/ 2.76 (-18.33/ 1.86 unaccelerated)}
D, [2024-08-23T23:25:32.070038 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.843s	3 -28.71/ 3.56 (-18.55/ 2.30 unaccelerated)}
D, [2024-08-23T23:25:32.077623 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.851s	3 -29.21/ 4.63 (-18.67/ 2.96 unaccelerated)}
D, [2024-08-23T23:25:32.091025 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +45.864s	3 -4.91/ 4.03 (-4.41/ 3.62 unaccelerated)}
D, [2024-08-23T23:25:32.109511 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_END       +45.883s	3}
D, [2024-08-23T23:25:32.814536 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_BEGIN     +46.587s	3}
D, [2024-08-23T23:25:32.814989 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +46.587s	3 17.54/-0.66 (17.54/-0.66 unaccelerated)}
D, [2024-08-23T23:25:32.821666 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +46.595s	3 25.82/-0.26 (21.38/-0.22 unaccelerated)}
I, [2024-08-23T23:25:32.822353 #149594]  INFO -- : {:command=>"xdotool key alt+Left", :args=>{:move_x=>25.82, :move_y=>-0.26, :unaccelerated_x=>21.38, :unaccelerated_y=>-0.22, :zoom=>0.0, :rotate=>0.0}}
D, [2024-08-23T23:25:32.829268 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +46.602s	3 44.96/ 0.64 (23.08/ 0.33 unaccelerated)}
D, [2024-08-23T23:25:32.836744 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE    +46.610s	3 52.45/ 2.55 (24.78/ 1.21 unaccelerated)}
[... cut ...]
D, [2024-08-23T23:25:38.848721 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_BEGIN      +52.622s	1}
D, [2024-08-23T23:25:38.870416 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event4   GESTURE_HOLD_END        +52.644s	1}
D, [2024-08-23T23:25:39.278018 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event2   KEYBOARD_KEY            +53.051s	*** (-1) pressed}
D, [2024-08-23T23:25:39.433257 #149594] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event2   KEYBOARD_KEY            +53.206s	*** (-1) pressed}

```

iberianpig, 2024-08-24:
Could you paste the result of `fusuma --version` ?

iberianpig, 2024-08-24:
This is the default config file.
https://github.com/iberianpig/fusuma/blob/main/lib%2Ffusuma%2Fconfig.yml
If the user's configuration file is not found, this file should be used.

Shantanu2k19, 2024-08-24:
```python

$ fusuma --version
I, [2024-08-24T21:53:38.189874 #12450]  INFO -- : reload config: /home/zodiac/.config/fusuma/config.yml
I, [2024-08-24T21:53:38.189907 #12450]  INFO -- : ---------------------------------------------
I, [2024-08-24T21:53:38.190039 #12450]  INFO -- : Fusuma: 3.6.1
I, [2024-08-24T21:53:38.190923 #12450]  INFO -- : libinput: 1.25.0
I, [2024-08-24T21:53:38.190948 #12450]  INFO -- : ruby 3.2.3p157
I, [2024-08-24T21:53:38.191829 #12450]  INFO -- : OS: Linux 6.8.0-41-generic #41-Ubuntu SMP PREEMPT_DYNAMIC Fri Aug  2 20:41:06 UTC 2024
I, [2024-08-24T21:53:38.192614 #12450]  INFO -- : Distribution: Ubuntu 24.04 LTS \n \l
I, [2024-08-24T21:53:38.193235 #12450]  INFO -- : Desktop session: ubuntu x11
I, [2024-08-24T21:53:38.193250 #12450]  INFO -- : ---------------------------------------------
I, [2024-08-24T21:53:38.193265 #12450]  INFO -- : Enabled Plugins: 
I, [2024-08-24T21:53:38.193322 #12450]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2024-08-24T21:53:38.193344 #12450]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2024-08-24T21:53:38.193352 #12450]  INFO -- :   Fusuma::Plugin::Detectors::HoldDetector
I, [2024-08-24T21:53:38.193359 #12450]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2024-08-24T21:53:38.193366 #12450]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2024-08-24T21:53:38.193373 #12450]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2024-08-24T21:53:38.193380 #12450]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2024-08-24T21:53:38.193387 #12450]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2024-08-24T21:53:38.193393 #12450]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2024-08-24T21:53:38.193400 #12450]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2024-08-24T21:53:38.193407 #12450]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2024-08-24T21:53:38.193414 #12450]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2024-08-24T21:53:38.193421 #12450]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2024-08-24T21:53:38.193429 #12450]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2024-08-24T21:53:38.193446 #12450]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2024-08-24T21:53:38.193459 #12450]  INFO -- : ---------------------------------------------
```

Thank-you for the answer.
How can I enable or disable the default values?

iberianpig, 2024-08-24:
With the logs below, the program seems to have correctly read your configuration file.

```
fusuma --version
I, [2024-08-24T21:53:38.189874 #12450]  INFO -- : reload config: /home/zodiac/.config/fusuma/config.yml
I, [2024-08-24T21:53:38.189907 #12450]  INFO -- : ----
```

Is there any problem when executing `$ fusuma` from terminal?

Shantanu2k19, 2024-08-24:
The problem is, fusuma is taking gestures that I did not define in my config file. I want to enable only the gestures that I defined and disable the default gestures.

iberianpig, 2024-08-25:
If the default gesture configuration file is loaded, the path should be displayed in the log. Kill the running Fusuma process once.
After that, would you like to see which path of config file loaded in the log when you run the fusuma command from terminal?

Shantanu2k19, 2024-08-25:
> Kill the running Fusuma process once.

This worked, Thanks

closed 2024-08-25 by Shantanu2k19

Shantanu2k19, 2024-08-25:
I also had touchegg installed. That seems to be the reason.