iberianpig/fusuma #324: Fusuma does not work with libinput 1.27.0-1 (Arch Linux and Hyprland)

juanfranUam, 2024-11-20:
Fusuma: 3.6.2

Fusuma does not recognize any gesture when I update to last version of libinput, if I downgrade to libinput 1.26.2-1 fusuma works again

**To Reproduce**
Steps to reproduce the behavior:
1. Install last version of libinput
2. Starts fusuma
3. Nothing happens

**Configuration**
swipe:
  3:
    left:
      command: "hyprctl dispatch workspace r+1" 
    right:
      command: "hyprctl dispatch workspace r-1" 
    up:
      command: "special_off.sh"
    down:
      command: "special_on.sh"

  4:
    left:
      command: "hyprctl dispatch movetoworkspace r+1"
    right:
      command: "hyprctl dispatch movetoworkspace r-1"
    up:
      command: "hyprctl dispatch movetoworkspace $(hyprctl monitors | grep active | awk '{print $3}')"
    down:
      command: "hyprctl dispatch movetoworkspace special:magic"

pinch:
  3:
    in:
      command: "hyprctl dispatch fullscreenstate 0"
    out:
      command: "hyprctl dispatch fullscreenstate 1"
  4:
    in:
      command: "hyprctl dispatch fullscreenstate 0"
    out:
      command: "hyprctl dispatch fullscreenstate 2"

iberianpig, 2024-11-21:
@juanfranUam 
Thanks for your reporting.
I have confirmed that this issue occurs with libinput 1.27.x. The cause is the change in the format of libinput `debug-events`. Please use libinput 1.26.x until an update is released.

J053Fabi0, 2024-11-22:
I face the same issue. In my Thinkpad X1 Yoga I find that moving the pointer with the trackpoint while performing the gestures on the touchpad allows fusuma to recognize them.

closed 2024-11-24 by iberianpig

iberianpig, 2024-11-24:
Support for libinput 1.27.0 has been added. Please try updating Fusuma.

Attacktive, 2024-11-25:
Thanks for the quick fix, @iberianpig. `fusuma 3.7.0` works with `libinput 1.27.0`.

PR #325 (2024-11-24): Support libinput 1.27.0 or later
This PR addresses compatibility with the `libinput debug-events` format introduced in libinput version 1.27.0.

Changes:

- Added a method `libinput_1_27_0_or_later?` in `libinput_command.rb` to determine if the libinput version is 1.27.0 or later.
- Updated `libinput_gesture_parser.rb` to handle the parsing of gesture events for libinput version 1.27.0 and later.

closes: #324