iberianpig/fusuma #335: Can fusuma use flatpak -> `device=input|all` instead of requiring `input` group?

KAGEYAM4, 2025-01-20:
**Is your feature request related to a problem? Please describe.**
> IMPORTANT: This makes /dev/input/ readable, so if that's an issue for you for some reason (like for privacy- or securityconcerns etc. or if it causes other parts of your OS to misbehave), consider this your heads-up.

**Describe the solution you'd like**
Can flatpak solve this ? https://docs.flatpak.org/en/latest/sandbox-permissions.html#device-access

iberianpig, 2025-01-26:
I'm not very familiar with Flatpak, but for Fusuma, if we can control access to `/dev/input`, we might not need the input group for libinput.
Also, fusuma-plugin-remap uses `/dev/uinput` to create virtual devices, which needs extra permissions.

Projects that use Flatpak, like Steam or other key remappers, might be useful to look at.

It could also be useful to see if we can run without the input group by using systemd to run as root.

KAGEYAM4, 2025-01-26:
> a Flatpak app runs as your user, so at best it can do what your user can do. --device=input merely provides access to the device nodes.

I asked on flatpak-Matrix, and flatpak solution won't work.

KAGEYAM4, 2025-01-28:
@iberianpig  found it on how steam-input works --> (TLDR: udev-rule granting tag:uaccess)
- https://www.reddit.com/r/archlinux/comments/viss0k/having_steam_installed_gives_your_user_readwrite/
- https://wiki.archlinux.org/title/Gamepad#Device_permissions
- https://github.com/ValveSoftware/steam-devices/blob/master/60-steam-input.rules

iberianpig, 2026-10-03:
The discussion confirms that Flatpak device permissions do not replace the host permissions needed to read  and create ; they only expose device nodes to the sandbox. The more promising direction is targeted udev rules, which is separate from Flatpak packaging. Closing this Flatpak-specific request as not planned. A focused issue for udev-based device permissions can be opened if someone wants to work on it.

closed 2026-10-03 by iberianpig