iberianpig/fusuma #341: Fusuma doesn't detect touchpad when mouse plugged in

auspisses, 2025-03-14:
**Describe the bug**
Fusuma doesn't detect touchpad device when mouse is plugged in

**To Reproduce**
Steps to reproduce the behavior:
1. Unplug mouse, run `fusuma` (works fine) or run `fusuma -l`, which outputs: 
```
I, [2025-03-14T18:43:02.089505 #18614]  INFO -- : reload config: /home/austin/.config/fusuma/config.yml
I, [2025-03-14T18:43:02.089624 #18614]  INFO -- : ---------------------------------------------
I, [2025-03-14T18:43:02.092348 #18614]  INFO -- : Fusuma: 3.7.0
I, [2025-03-14T18:43:02.094658 #18614]  INFO -- : libinput: 1.27.1
I, [2025-03-14T18:43:02.094728 #18614]  INFO -- : ruby 3.3.7p123
I, [2025-03-14T18:43:02.096600 #18614]  INFO -- : OS: Linux 6.13.6-200.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Mar  7 21:33:48 UTC 2025
I, [2025-03-14T18:43:02.098798 #18614]  INFO -- : Distribution: \S
Kernel \r on an \m (\l)
I, [2025-03-14T18:43:02.102088 #18614]  INFO -- : Desktop session: gnome wayland
I, [2025-03-14T18:43:02.102156 #18614]  INFO -- : ---------------------------------------------
I, [2025-03-14T18:43:02.102182 #18614]  INFO -- : Enabled Plugins: 
I, [2025-03-14T18:43:02.102288 #18614]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2025-03-14T18:43:02.102313 #18614]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2025-03-14T18:43:02.102336 #18614]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2025-03-14T18:43:02.102352 #18614]  INFO -- :   Fusuma::Plugin::Detectors::HoldDetector
I, [2025-03-14T18:43:02.102371 #18614]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2025-03-14T18:43:02.102387 #18614]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2025-03-14T18:43:02.102400 #18614]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2025-03-14T18:43:02.102415 #18614]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2025-03-14T18:43:02.102433 #18614]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2025-03-14T18:43:02.102449 #18614]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2025-03-14T18:43:02.102488 #18614]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2025-03-14T18:43:02.102511 #18614]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2025-03-14T18:43:02.102532 #18614]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2025-03-14T18:43:02.102556 #18614]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2025-03-14T18:43:02.102575 #18614]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2025-03-14T18:43:02.102590 #18614]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2025-03-14T18:43:02.102607 #18614]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2025-03-14T18:43:02.102623 #18614]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2025-03-14T18:43:02.102636 #18614]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2025-03-14T18:43:02.102654 #18614]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2025-03-14T18:43:02.102670 #18614]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2025-03-14T18:43:02.102685 #18614]  INFO -- : ---------------------------------------------
ELAN0678:00 04F3:3195 Touchpad
```
2. Kill Fusuma if already running. (Fusuma works if mouse is plugged in after Fusuma has been started; it should regardless of when mouse is plugged in.) Running `fusuma` outputs following (backticks are bracketed for formatting purposes):
```
I, [2025-03-14T18:45:14.862428 #19911]  INFO -- : reload config: /home/austin/.config/fusuma/config.yml
I, [2025-03-14T18:45:14.862476 #19911]  INFO -- : ---------------------------------------------
I, [2025-03-14T18:45:14.863452 #19911]  INFO -- : Fusuma: 3.7.0
I, [2025-03-14T18:45:14.864617 #19911]  INFO -- : libinput: 1.27.1
I, [2025-03-14T18:45:14.864639 #19911]  INFO -- : ruby 3.3.7p123
I, [2025-03-14T18:45:14.865631 #19911]  INFO -- : OS: Linux 6.13.6-200.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Mar  7 21:33:48 UTC 2025
I, [2025-03-14T18:45:14.866345 #19911]  INFO -- : Distribution: \S
Kernel \r on an \m (\l)
I, [2025-03-14T18:45:14.867906 #19911]  INFO -- : Desktop session: gnome wayland
I, [2025-03-14T18:45:14.867923 #19911]  INFO -- : ---------------------------------------------
I, [2025-03-14T18:45:14.867933 #19911]  INFO -- : Enabled Plugins: 
I, [2025-03-14T18:45:14.867982 #19911]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2025-03-14T18:45:14.867990 #19911]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2025-03-14T18:45:14.867997 #19911]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2025-03-14T18:45:14.868002 #19911]  INFO -- :   Fusuma::Plugin::Detectors::HoldDetector
I, [2025-03-14T18:45:14.868008 #19911]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2025-03-14T18:45:14.868013 #19911]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2025-03-14T18:45:14.868020 #19911]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2025-03-14T18:45:14.868025 #19911]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2025-03-14T18:45:14.868030 #19911]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2025-03-14T18:45:14.868036 #19911]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2025-03-14T18:45:14.868042 #19911]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2025-03-14T18:45:14.868048 #19911]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2025-03-14T18:45:14.868054 #19911]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2025-03-14T18:45:14.868059 #19911]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2025-03-14T18:45:14.868067 #19911]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
[... 121 lines cut ...]
          sendkey: "LEFTMETA+DOWN"

rotate:
  3:
    counterclockwise:
      sendkey: "LEFTMETA+LEFTSHIFT+PAGEUP"
    clockwise:
      sendkey: "LEFTMETA+LEFTSHIFT+PAGEDOWN"

plugin:
  filters:
    libinput_device_filter:
      keep_device_names:
        - "ELAN0678:00 04F3:3195 Touchpad"
  executors:
    sendkey_executor:
      device_name: 'AT Translated Set 2 keyboard'
```

Note: Removing the `filters` plugin makes `fusuma -l` output mouse devices at the end:
```
I, [2025-03-14T18:50:35.873693 #22521]  INFO -- : ---------------------------------------------
HP, Inc HyperX Pulsefire Haste
HP, Inc HyperX Pulsefire Haste Consumer Control
```

iberianpig, 2025-04-02:
@auspisses 
Could you please provide the output of the following commands:

- The output of "libinput list-devices" when a mouse is connected.
- The output of "libinput list-devices" when no mouse is connected.

auspisses, 2025-04-06:
**Mouse connected**
```
libinput-config: initializing
libinput-config: replacing the device config functions
libinput-config: initializing the key mapper
libinput-config: option 'scroll-factor' is '0.5'
libinput-config: initialized
libinput-config: configuring device 'Power Button'
Device:           Power Button
Kernel:           /dev/input/event3
Group:            1
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   n/a
Rotation:         0.0

libinput-config: configuring device 'Video Bus'
Device:           Video Bus
Kernel:           /dev/input/event9
Group:            2
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   n/a
Rotation:         0.0

libinput-config: configuring device 'Power Button'
Device:           Power Button
Kernel:           /dev/input/event0
Group:            3
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   n/a
Rotation:         0.0

libinput-config: configuring device 'Lid Switch'
Device:           Lid Switch
Kernel:           /dev/input/event1
[... 563 lines cut ...]
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   n/a
Rotation:         0.0

libinput-config: configuring device 'keyd virtual pointer'
Device:           keyd virtual pointer
Kernel:           /dev/input/event21
Group:            13
Seat:             seat0, default
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      identity matrix
Scroll methods:   button
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   flat *adaptive custom
Rotation:         0.0
```

iberianpig, 2025-04-07:
I've reviewed the information you provided, but I wasn’t able to reproduce the issue on my end, so further investigation is quite challenging at this point.

One thing I’m not entirely clear on is why the touchpad device becomes unrecognized when a mouse is connected. Do you have any idea or any insight into this behavior? Also, I'm not familiar with using keyd; if you try disabling keyd, does the behavior change?

Any additional observations you can share would be greatly appreciated. Thank you!

auspisses, 2025-04-10:
Regarding keyd: I disabled its systemd service via `systemctl stop keyd.service`, which makes keyd's virtual devices disappear from `libinput list-devices`. (Might not be the best way to disable keyd but I'm not super Linux savvy yet). Running `fusuma -l` without mouse, and it detects my touchpad; with mouse, it doesn't and only detects the mouse. 

With no mouse, when I run `fusuma -v` and tap my touchpad, this is one of the lines in the output
```
D, [2025-04-10T15:43:14.389999 #166339] DEBUG -- : {:available_devices=>[#<Fusuma::Device:0x00007fe4d1b99ca0 @id="event7", @name="ELAN0678:00 04F3:3195 Touchpad", @capabilities="pointer gesture", @available=true>, #<Fusuma::Device:0x00007fe4d1b9b730 @id="event6", @name="ELAN0678:00 04F3:3195 Mouse", @capabilities="pointer", @available=true>, #<Fusuma::Device:0x00007fe4d1b966e0 @id="event5", @name="TPPS/2 Elan TrackPoint", @capabilities="pointer", @available=true>]}
```
(I had commented out the plugin line that keeps only the touchpad device, so some mouse device (none are hooked up to my system) and my trackpoint are detected. Leaving the comment in makes fusuma work w/o mouse and the device not found by fusuma w/ mouse.)

Here's a part of `fusuma -v` when the mouse is plugged in (and I tap my touchpad so the debug shows available devices):
```
D, [2025-04-10T15:40:16.483643 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event20  DEVICE_ADDED                HP, Inc HyperX Pulsefire Haste    seat0 default group6  cap:p left scroll-nat scroll-button}
D, [2025-04-10T15:40:16.483715 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event21  DEVICE_ADDED                HP, Inc HyperX Pulsefire Haste System Control seat0 default group6  cap:k}
D, [2025-04-10T15:40:16.483793 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event22  DEVICE_ADDED                HP, Inc HyperX Pulsefire Haste Consumer Control seat0 default group6  cap:kp scroll-nat}
D, [2025-04-10T15:40:16.483872 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event24  DEVICE_ADDED                HP, Inc HyperX Pulsefire Haste    seat0 default group6  cap:k}
D, [2025-04-10T15:40:16.483934 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event19  DEVICE_ADDED                DCMT Technology USB Condenser Microphone seat0 default group7  cap:k}
D, [2025-04-10T15:40:16.484262 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event9   DEVICE_ADDED                ELAN901C:00 04F3:2EDE             seat0 default group8  cap:t  size 317x197mm ntouches 10 calib}
D, [2025-04-10T15:40:16.484553 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event6   DEVICE_ADDED                ELAN0678:00 04F3:3195 Mouse       seat0 default group9  cap:p left scroll-nat scroll-button}
D, [2025-04-10T15:40:16.484816 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event7   DEVICE_ADDED                ELAN0678:00 04F3:3195 Touchpad    seat0 default group9  cap:pg  size 114x58mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on}
D, [2025-04-10T15:40:16.484890 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event4   DEVICE_ADDED                AT Translated Set 2 keyboard      seat0 default group10 cap:k}
D, [2025-04-10T15:40:16.484953 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event5   DEVICE_ADDED                TPPS/2 Elan TrackPoint            seat0 default group11 cap:p left scroll-nat scroll-button}
D, [2025-04-10T15:40:16.485204 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event13  DEVICE_ADDED                ThinkPad Extra Buttons            seat0 default group12 cap:k}
D, [2025-04-10T15:40:16.485338 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event27  DEVICE_ADDED                JLab JBuds Mini (AVRCP)           seat0 default group13 cap:k}
D, [2025-04-10T15:40:17.946861 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record: -event7   GESTURE_HOLD_BEGIN          +0.040s	1}
D, [2025-04-10T15:40:17.948637 #164786] DEBUG -- : {:list_devices=>"libinput list-devices"}
D, [2025-04-10T15:40:18.347899 #164786] DEBUG -- : {:available_devices=>[#<Fusuma::Device:0x00007f2736516058 @id="event20", @name="HP, Inc HyperX Pulsefire Haste", @capabilities="pointer", @available=true>, #<Fusuma::Device:0x00007f2736513538 @id="event22", @name="HP, Inc HyperX Pulsefire Haste Consumer Control", @capabilities="keyboard pointer", @available=true>]}
D, [2025-04-10T15:40:18.348132 #164786] DEBUG -- : {:input_event=>tag: libinput_command_input, record:  event7   GESTURE_HOLD_END            +0.334s	1}
```
It appears that some aspect of how fusuma figures out available devices prevents the touchpad from being available when the mouse is available. I also tried fusuma with a friend's mouse and it didn't work, so it doesn't appear to be because of my particular mouse.

phuchoang2603, 2025-04-12:
i actually have the same issues. i have been trying to set up `fusuma-plugin-sendkey` but whenever i am trying to list the devices, it always return like this. plug out the mouse helped me solve it.
```
$ fusuma-sendkey -l
sendkey: Keyboard: /keyboard|Keyboard|KEYBOARD/ is not found
```

iberianpig, 2025-04-12:
I noticed that the logs include output from libinput‑config(https://git.samuelcollins.dev/sam/libinput-config) .

Upon testing libinput-config, I was able to reproduce the same results as reported when using libinput‑config. 
Could you confirm if you are using libinput‑config in your environment?

closed 2025-04-12 by iberianpig

iberianpig, 2025-04-12:
Fusuma v3.8.1 has been released. Please update to the latest version and check if the problem has been resolved. If you still encounter any issues, feel free to reopen this issue.

Thank you!

auspisses, 2025-04-13:
Yes I am using libinput-config. After updating to 3.8.1, seems like Fusuma is working perfectly! Thank you very much!

commit 8ffa8dd6 (2025-04-12): fix: avoid stdout block by reading stderr output

Open3.popen3 outputs both stdout and stderr continuously.
When using libinput-config, debug logs are sent to stderr.
Not reading stderr caused stdout to block, leading to
unexpected output. This commit addresses the issue.

Close #341