iberianpig/fusuma #342: Fusuma does not work with libinput 1.28.0 on archlinux

cheanus, 2025-03-29:
**Describe the bug**
When using libinput 1.28.0, any pinch gesture does not work. if downgrade to libinput 1.27, fusuma works again.

**To Reproduce**
Steps to reproduce the behavior:
1. Update libinput to 1.28.0 and start fusuma
2. Add a pinch gesture in config.yml of fusuma
3. Pinch gesture does not work.

**Expected behavior**
Pinch gesture can work.

**Versions**
```yml
I, [2025-03-29T22:50:06.694390 #4529]  INFO -- : reload config: /home/test/.config/fusuma/config.yml
I, [2025-03-29T22:50:06.694434 #4529]  INFO -- : ---------------------------------------------
I, [2025-03-29T22:50:06.694581 #4529]  INFO -- : Fusuma: 3.7.0
I, [2025-03-29T22:50:06.695151 #4529]  INFO -- : libinput: 1.28.0
I, [2025-03-29T22:50:06.695199 #4529]  INFO -- : ruby 3.3.7p123
I, [2025-03-29T22:50:06.695706 #4529]  INFO -- : OS: Linux 6.13.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 23 Mar 2025 17:17:30 +0000
I, [2025-03-29T22:50:06.696129 #4529]  INFO -- : Distribution: \S{PRETTY_NAME} \r (\l)
I, [2025-03-29T22:50:06.696933 #4529]  INFO -- : Desktop session: plasma wayland
I, [2025-03-29T22:50:06.696942 #4529]  INFO -- : ---------------------------------------------
I, [2025-03-29T22:50:06.696951 #4529]  INFO -- : Enabled Plugins: 
I, [2025-03-29T22:50:06.696995 #4529]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2025-03-29T22:50:06.697002 #4529]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2025-03-29T22:50:06.697009 #4529]  INFO -- :   Fusuma::Plugin::Detectors::HoldDetector
I, [2025-03-29T22:50:06.697015 #4529]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2025-03-29T22:50:06.697021 #4529]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2025-03-29T22:50:06.697028 #4529]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2025-03-29T22:50:06.697034 #4529]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2025-03-29T22:50:06.697040 #4529]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2025-03-29T22:50:06.697046 #4529]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2025-03-29T22:50:06.697052 #4529]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2025-03-29T22:50:06.697058 #4529]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2025-03-29T22:50:06.697064 #4529]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2025-03-29T22:50:06.697070 #4529]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2025-03-29T22:50:06.697076 #4529]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2025-03-29T22:50:06.697082 #4529]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2025-03-29T22:50:06.697089 #4529]  INFO -- : ---------------------------------------------
```

**Configuration**
```yml
swipe:
  3:
    up:
      command: ydotool key 125:1 29:1 56:1 103:1 103:0 56:0 29:0 125:0 # Toggle Maximize/Unmaximize Window
    down:
      command: ydotool key 125:1 29:1 56:1 108:1 108:0 56:0 29:0 125:0 # Minimize window
    # begin:
    #   command: ydotool key 56:1
    # right:
    #   update:
    #     command: ydotool key 15:1 15:0
    #     interval: 5
    # left:
    #   update:
    #     command: ydotool key 42:1 15:1 15:0 42:0
    #     interval: 5
    # end:
    #   command: ydotool key 56:0
pinch:
  # 2:
  #   in:
  #     begin:
  #       command: ydotool key 125:1
  #     update:
  #       command: ydotool key 12:1 12:0
  #       interval: 0.75
  #     end:
  #       command: ydotool key 125:0 # Zoom in
  #   out:
  #     begin:
  #       command: ydotool key 125:1
  #     update:
  #       command: ydotool key 13:1 13:0
  #       interval: 0.75
  #     end:
  #       command: ydotool key 125:0 # Zoom out
  3:
    in:
      interval: 1
      threshold: 0.8
      command: ydotool key 56:1 62:1 62:0 56:0 # close window
    out:
      command: ydotool key 125:1 32:1 32:0 125:0 # Switch to desktop
  4:
    in:
      command: ydotool key 29:1 56:1 38:1 38:0 56:0 29:0 # Open/close yakuake

hold:
  4:
    command: ydotool key 29:1 56:1 34:1 34:0 56:1 29:1 # Notice

threshold:
  pinch: 0.5
```

rien333, 2025-03-29:
Seeing the same thing, also only on `libinput` 1.28, and not on 1.27. 

Maybe useless information, but these are my input devices:

```
$ libinput list-devices | grep 'Device'
Device:           Video Bus
Device:           Lid Switch
Device:           Power Button
Device:           FRMW0004:00 32AC:0006 Wireless Radio Control
Device:           FRMW0004:00 32AC:0006 Consumer Control
Device:           PIXA3854:00 093A:0274 Touchpad
Device:           PIXA3854:00 093A:0274 Mouse
Device:           AT Translated Set 2 keyboard
```

If I `fusuma-sendkey` to `AT Translated Set 2 keyboard` on 1.28, I get `sendkey: Keyboard: /AT Translated Set 2 keyboard/ is not found`.  Which is strange, since it's definitely present in the output of `libinput list-devices`

On 1.27, however, the key is send fine, and no errors are shown.

iberianpig, 2025-04-01:
This issue has been resolved in libinput 1.28.1.

Please update to 1.28.1 and verify that the pinch gesture is working as expected.
If you have any issues, please reopen.

closed 2025-04-01 by iberianpig

rien333, 2025-04-03:
Still present on 1.28.1 for me. Symptoms are the same as the ones I described earlier in this thread: https://github.com/iberianpig/fusuma/issues/342#issuecomment-2764284065